flipperdevices/flipperzero-firmware #1182: Frequency Analyzer displays signal when there is none

ToddNutty, 2022-04-28:
**Describe the bug**
After Flipper has scanned 315MHz Frequency Analyzer will display signal when there is none. Signal appears to range from 299.999 - 301.000 with RSSI often showing no to very low signal.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on...
2. Enter Sub-GHz
3. Use Read function
4. Choose Config
5. Set Frequency to: 315 / Hopping: OFF / Modulation: AM650
6. Allow Flipper to scan for ~10 seconds
7. Exit
8. Open Frequency Analyzer
9. Display should show 299.999  to ~301.000 **(May need to physically move Flipper around)**

**Example screen capture: https://imgur.com/a/EjpFBg0**

**Expected behavior**
Frequency Analyzer should display all 0's (000.000) and only display greater values when signal is detected

**Additional context**
The problem only seems to occur after using the Read function within Sub-GHz and scanning on the 315 frequency for a short time.
[image]
[image]

Skorpionm, 2022-04-29:
congratulations, next to you something very strong is broadcasting on this frequency.
the frequency analyzer will still determine the key fobs, since they should be placed closer to the flipper. i can lower the RSSI detection level by this will lower the sensitivity

ToddNutty, 2022-04-29:
> congratulations, next to you something very strong is broadcasting on this frequency. the frequency analyzer will still determine the key fobs, since they should be placed closer to the flipper. i can lower the RSSI detection level by this will lower the sensitivity

I generally don't have issue with just restarting the Flipper, but what is interesting is I don't run into this issue unless I've previously tried scanning on 315Mhz on Read. I'll test at another location much further away to rule out local background noise.

jimilinuxguy, 2022-05-01:
I wanted to tag on that I have seen this several times as well. Either the Frequency Analyzer will show a signal with 1-2 bars of RSSI or if you are recording, it will show actual graph/wave bars but its not real. I have checked with RTL-SDR and HackRF at the same time that I saw it. Restarting the device typically has made the error go away.

Astrrra, 2022-05-03:
I have the same behavior in a basically radio-silent location

Skorpionm, 2022-05-03:
I have an idea, I'll check it out tonight

Skorpionm, 2022-05-03:
fix https://github.com/flipperdevices/flipperzero-firmware/pull/1192

closed 2022-05-07 by Skorpionm