flipperdevices/flipperzero-firmware #1186: Automatic desktop locking with PIN code

digitalcircuit, 2022-04-29:
**Scenario**
The [automatic desktop locking feature](https://github.com/flipperdevices/flipperzero-firmware/pull/1101 ) works well for avoiding accidental button presses when in one's pocket, etc.  However, it's not very effective at reducing risk if someone else finds the device (lost, etc).

**Describe the solution you'd like**
An additional setting (e.g. `Settings` → `Desktop` → `PIN Setup` → `PIN Lock After`) that locks the Flipper with one's PIN code after a configurable delay.

For example, with `Auto Lock Time: 60s` and `PIN Lock After: 10min`, the Flipper will use the triple-back lock after 60 seconds, then after 10 minutes, it'll lock with the PIN Code (also disabling USB).

**Describe alternatives you've considered**
* Replacing (instead of augmenting) the triple-back lock with PIN code lock
   * In `Settings` → `Desktop` → `PIN Setup` → `Auto Lock w/PIN` as an `ON`/`OFF`, or similar
   * ℹ This might make more sense (especially if auto lock will only ever work while viewing the desktop)
   * ➕ Simple, straightforward
   * ➖ Makes lock time setting a trade-off between guarding against accidental button presses and time it takes to input one's PIN code

**Additional context**
To try to reduce confusion, `PIN Lock After` should probably not be allowed to be set to to a shorter delay than `Auto Lock Time` unless `Auto Lock Time` is set to `OFF`.

It might make more sense to move these settings out to a `Security` category if the screen locking feature can be applied system-wide (instead of only happening when on the desktop).  Or, `PIN Lock After` may be placed beneath `Auto Lock Time` so it's possible to see and control both automatic locking options from the same place.

I realize the Flipper's not designed to be completely secure against physical access, but I think this might help with situations where it's been left or lost and someone with good intentions wishes to return the device.

A separate feature request might be optional text to show on the lock screen (e.g. `Found: 123-555-6789`), similar to how Android allows specifying a lock screen message - I can file a new issue for that.

zhovner, 2022-05-08:
Thanks, will think about this.

theboojit, 2022-05-13:
Hi, 

I was going to take a stab at this just for fun, but I have some questions. 

Most importantly, does anyone see any reason why we shouldn't lock with PIN as the default behavior, if the user has both PIN set and auto-lock set? 

I was first approaching this as adding an extra time value for PIN auto-lock as @digitalcircuit describes above, but in looking at the implementation, it does complicate things. For example, ideally the value shouldn't be configurable if the user doesn't have a PIN, and you need to figure out what to do if the user creates a pin and then deletes it. Also, should a user be allowed to set the regular auto-lock time longer than the PIN auto-lock time? All these things are totally do-able but complicate the implementation unnecessarily.

For me, it seems like if I've set a PIN and I've enabled auto-lock, I really want the damn thing to lock with the PIN. From an implemenation standpoint and from a clear UX standpoint, does anyone disagree that we should just not add any more configuration here, and instead just implement "if the user has a PIN set and auto-lock set, then always lock with PIN when auto-lock timeout hits"?

digitalcircuit, 2022-05-14:
@theboojit Especially considering the auto-lock doesn't happen unless you're on the desktop view (which surprised me), I'd be happy with your proposed choice ("if PIN set and auto-lock set, always lock with PIN").

Later on, "DUMB mode" could be a PIN lock feature, wherein if you have a PIN lock set, you could opt-in to entering the dumb mode when locking with PIN, e.g. `Settings` → `Desktop` → `PIN Setup` → `Disguise on Lock` (or whatever dumb mode gets called officially).  I'm not sure if it'd make much sense to use dumb mode without a PIN in place.

skotopes, 2022-05-16:
We'll redesign locking in future updates. Thanks for feedback, that really helps.

smartynov, 2022-11-25:
Any updates on this? Would really appreciate this feature (especially the "dumb mode" auto lock)

skotopes, 2022-12-01:
@smartynov everyone is busy, we are planning to return to this issue closer to v1.0.0 release. Or maybe someone from community will implement it.

Swivelgames, 2023-05-16:
Was this also fixed by #2665?

skotopes, 2023-06-22:
I think we've implemented everything we wanted and this issue can be closed.

closed 2023-06-22 by skotopes