flipperdevices/flipperzero-firmware #1187: Save Sub-GHz keys to microSD once found

JulyIghor, 2022-05-02:
While using constant monitoring mode, it captures a keys.
The problem is if you forget to save them or if battery drain out they will be lost.
Keys need to be saved, at least in some temporary folder, once found.

RogueMaster, 2022-05-07:
can raw_signal saves have a datetime stamp added? Also, would it be possible for Frequency Analyzer to save data it found?

Skorpionm, 2022-05-12:
the frequency analyzer works only with devices that you bring close to the flipper.  and does not catch devices more than 15 centimeters away from it.  this was done on purpose so as not to catch strong signals from the air, and the flipper showed directly the frequency of the device in the hands

skotopes, 2022-12-01:
I do understand use case, but it's kinda shady one. I'll close this ticket.

closed 2022-12-01 by skotopes