flipperdevices/flipperzero-firmware #1188: Send Sub-GHz keys to mobile app once captured and store its GPS coordinates

JulyIghor, 2022-05-02:
If you use Sub-GHz constant monitoring, you may capture a lot of keys while walking.
But you don't know where is the keys located.
I suggest to push the keys over BlueTooth to mobile app once they found.
So mobile app can store GPS coordinates and time and it become more useful.

skotopes, 2022-05-16:
Please open this feature request in mobile apps

closed 2022-05-16 by skotopes

Stichoza, 2025-07-14:
This could also be possible through GPS module connected to GPIO.

BTW, if you add location data to your .sub files, they could be displayed on Flipper Map web app:

- [flipper-map.stichoza.com](https://flipper-map.stichoza.com/)

Looks like this:

[image]