flipperdevices/flipperzero-firmware #1201: SD card Encryption

iConfuzzled, 2022-05-05:
With the SD card storing all of the data such as our keyfobs, NFC cards, remotes, & etc, I'm wondering if it is possible to encrypt the data on the SD Card so that a simple eject will not leak all of our information.

I had asked this question on Discord, and Aleksandr Kutuzov from the flipper team replied: 

> Yes, it's possible to encrypt data on sd with unique private AES key from secure enclave

Since this seems to be possible, could this be added as a feature? Maybe even only enable the SD card encryption when user adds a lock pin to their Flipper.

Thanks!

RogueMaster, 2022-05-07:
I too think there should be a way to encrypt the keys/files on the SD so only a flipper that has unlocked it with a special password can read the data. Locking the file with the same pin as OS would be a good idea. That way, unlocking the flipper with your PIN provides the flipper OS with the password to unencrypt the key data. Perhaps each file can get an extra option to have it converted to be stored as encrypted that way not all files are stored this way since transferred files would be unencrypted by default. The encrypted list could be stored the same way pinned files are stored.

RogueMaster, 2022-05-07:
Also, I have covered my microSD with black tape to make it less noticeable and harder to remove.

GMMan, 2022-05-09:
There already appears to be a framework for this with U2F cert encryption, so maybe it can be built on that

closed 2023-10-29 by hedger