flipperdevices/flipperzero-firmware #1203: Option to only turn on backlight when pressing a button if the backlight is off

GMMan, 2022-05-07:
**Is your feature request related to a problem? Please describe.**
When pressing a button while the backlight is off, the button activates whatever option the button is intended to do, and also turns on the backlight. However, most of the time I won't remember what screen the device was previously on, and I won't be able to tell under low light conditions. In this situation, activating any option may be undesirable.

**Describe the solution you'd like**
Add an option to ignore what a button press should activate and just turn on backlight. Any subsequent presses will work as usual.

**Describe alternatives you've considered**
N/A

zhovner, 2022-05-08:
There is no way to do this without braking normal UI. Try DOWN button for this.

closed 2022-05-08 by zhovner

GMMan, 2022-05-08:
Down button on home screen activates browser, which then requires a press of the back button to return to desktop, so it's still not really a no-op. By breaking normal UI, do you mean there's no good place to hook input, or too much work to refactor current input handling?

GMMan, 2022-05-08:
Taking a quick look through the code, how about firing an on key down event after debounce in the input service, which the notification service can listen for, and allow it to cancel whatever other events associated with the key down/key up event? Allowing input to be marked as handled and not propagated further is also a common function on other platforms.

zhovner, 2022-05-08:
This will add annoying actions when you see the screen, for example in daylight, and don't need to additionally active backlight.

GMMan, 2022-05-08:
Which is why I mentioned it should be an option (perhaps I should have been more clear that I wanted this as something configurable, not forced upon everyone). This would be great in my situation, and for those who don't want it, they can choose to leave it off.

RogueMaster, 2022-05-08:
Perhaps holding center for 5 seconds before subsequent commands would keep the backlight off? Then later after next time out it resumes normal operation.