flipperdevices/flipperzero-firmware #1208: Allow multiple keyboard mappings for non-us English keyboard for BadUSB

jaixan, 2022-05-08:
**Is your feature request related to a problem? Please describe.**
I have a macOS with a French-Canadian Keyboard. When I run the sample "demo_macos.txt" bad USB script, I get something like this :

cat " édevénull '' EOF
dquote> Hello World!
dquote> ============================================================
dquote> 
dquote>               _.-------.._                    -,
dquote>           .-ÈùùùÈ--..,,_é éù-,               -,  à 
dquote>        .:È          é:é  éèà  à     ,_...,  ù. À  À
dquote>       é       ,----é:é  éùà _àÙù_-Èù     _;
dquote>      è      é éùÈÈÈèà à à.Ùù_-è      ,-Èèé 
dquote>     À      À À  0    À À .-è      ,éù  é
dquote>    À    ,..à à     ,.-Èù       ,éù    é
dquote>   ;    :    ùéùÈÈàù           ,é--==,é-----,
dquote>   À    ù-...À        -.___-Z:_______J...---;
dquote>   :         ù                           _-è
dquote>  _L_  _     ___  ___  ___  ___  ____--Èù
dquote> À __ÀÀ À   À_ _ÀÀ _ àÀ _ àÀ __ÀÀ _ à 
dquote> À _À À À__  À À À  _éÀ  _éÀ _À À   é 
dquote> À_À  À____ÀÀ___ÀÀ_À  À_À  À___ÀÀ_À_à 
dquote> 
dquote> Flipper Zero BadUSB feature is compatible with USB Rubber Ducky script format
dquote> More information about script synax can be found here:
dquote> https:éégithub.coméhak5darrenéUSB-Rubber-DuckyéwikiéDuckyscript
dquote> EOF

The hid_asciimap is build around an US English keyboard. Therefore the wrong keys are "pressed" when the target computer has another keyboard layout.

**Describe the solution you'd like**
The best solution would be to support the DUCKY_LANG keyword, allowing to change on the fly the mapping of the keyboard.

**Describe alternatives you've considered**
Another solution is to allow in the settings to reconfigure the keyboard mappings.

Thanks!

skotopes, 2022-05-16:
I'll add this to back log, we'll come back to it closer to mid summer. @nminaylov

thefunkygibbon, 2022-07-15:
guess you already know but the same goes for all types of keyboards.  UK keyboards for example give a @ when it should be a "
and a few other discrepancies. 

shame that a product that was created in non-US and sold mostly to non-US has a bias towards US :-)

bbock, 2022-07-29:
for a 3rd party implementation of this feature see [here](https://github.com/v1nc/flipperzero-firmware).

gilou, 2022-08-17:
ah, I had a lot of fun on a Windows using FR BEPO, sharing its keyboard with another station on another mapping haha, the result is… interesting. Anyway, it would be great to be able to switch / target mapping without having to recompile indeed. Also I wonder if it’s possible to get that information from the victim system, to avoid having to gather that info beforehand…

Fun exercise would be to write payloads that work on as many mappings as possible :P

ahumeniy, 2022-09-29:
you can make scripts which works regardless of the keyboard mapping, but you have to convert all the strings to alt+ codes. A side effect of that being the script is no longer readable

skotopes, 2023-03-13:
Arrived couple releases ago

closed 2023-03-13 by skotopes