flipperdevices/flipperzero-firmware #1210: JTAG/SWD debug probe functionality

GMMan, 2022-05-09:
**Is your feature request related to a problem? Please describe.**
It would be nice if the device can act as a JTAG/SWD probe.

**Describe the solution you'd like**
It can be based off of other open source projects. Raspberry Pi's picoprobe looks simple enough, where the host controls the pin state. Support should be already present in Pi's fork of OpenOCD, though not sure if it's been upstreamed.

**Describe alternatives you've considered**
Other projects exist, such as CMSIS-DAP, or an emulation of ST-Link protocol (requires more work on the probe's end, but has its canonical implementation on a more common STM32, so should be feasible), though those projects aren't as simple as picoprobe.

**Additional context**
No level shifting support by the hardware itself presumably, but that can probably be addressed with an external level shifting board,

skotopes, 2022-05-16:
We have plans to port blackmagic. It's scheduled right after we add ability to load apps from SD.

skotopes, 2022-12-01:
Check DAP link app bundled with firmware

closed 2022-12-01 by skotopes