flipperdevices/flipperzero-firmware #1226: Music player plays back incorrect BPM

OnyxtheDragon, 2022-05-12:
**Describe the bug**
The music player appears to play back RTTTL files slightly faster than the BPM encoded in the file

**To Reproduce**
Steps to reproduce the behavior:
1. Play back song with 120 BPM
2. Adjust the BPM of the song down 15-18 BPM to get correct BPM

**Expected behavior**
Flipper plays back files at correct BPM 120-120 instead of 120-135

Attached is a music file i have created based on https://www.youtube.com/watch?v=Yw6u6YkTgQ4

[Hello World.txt](https://github.com/flipperdevices/flipperzero-firmware/files/8681592/Hello.World.txt)

skotopes, 2022-05-13:
```Hello World:d=4,o=5,b=120:5f5,5c5,1g5,3p,5a5,5c6,5a5,5f5,5d5,5c5,5d5,8a5,8g5,2a5,2p,5a5,5g5,5f5,5e5,5d5,5a4,5c5,8e5,8f5,2e5,2p```

5f5 - is 1/5th, F, octave 5. Are you sure that it should be 1/5th? 
Because if yes then it is correct value. But it's not correct size in terms of music.

Duration calculation looks like that:

```
60 * 1000 * 4 / 120 / 5 = 400ms
60 - seconds per minute
1000 ticks per second
4 - 4/4 musical size
120 - your bpm
5 - note size 1/5th
```

Also there is a small drift (up to 2ms per note), but that should be negligible.

OnyxtheDragon, 2022-05-13:
Its definitely possible that i'm in error here! I'll try to use a precise method of testing and see whats up.

closed 2022-05-16 by skotopes