flipperdevices/flipperzero-firmware #1230: Cannot compile firmware on Apple Silicon

adamm, 2022-05-14:
Hello,

Building Flipper Firmware on MacOS using Apple Silicon fails with a super strange syntax error messages.   In my case I have a Macbook Air (M1 2020).

TLDR: There is a workaround.  Downgraded brew cask gcc-arm-embedded to version 10.3.   See below.

**To Reproduce**
Steps to reproduce the behavior:
1. Get an Apple M1 laptop
2. `brew bundle --verbose`
3. `make`
4. Build starts OK, but soon very strange "syntax" errors soon occur.  Build falis.

**Logs**
I'll attach the full log output, but here's a snippet.  4 or 5 other errors with seemingly basic syntax similar to the following:
```
/Users/adamm/git/flipper/unleashed/applications/power/power_service/power.c: In function 'power_check_low_battery':
/Users/adamm/git/flipper/unleashed/applications/power/power_service/power.c:155:5: internal compiler error: Illegal instruction: 4
  155 |     if((power->info.charge == 0) && (power->info.voltage_vbus < 4.0f) &&
      |     ^~
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
make[1]: *** [.obj/f7-firmware//Users/adamm/git/flipper/unleashed/applications/power/power_service/power.o] Error 1
```

**Workaround**
Downgrade your brew installation of gcc-arm-ebedded to 10.3.
1. `cd $(brew --repo homebrew/cask)`
2. `git checkout d407663b8017a0a062c7fc0b929faf2e16abd1ff`    # version 10.3 of gcc-arm-embedded
3. `HOMEBREW_NO_AUTO_UPDATE=1 brew reinstall gcc-arm-embedded`
4. `git checkout master`
5. `make` in flipperzero-firmware now works.

**Additional context**
Issue occurs both on dev and release branches.

DrZlo13, 2022-05-16:
Same. GCC 11 on M1 is broken.

skotopes, 2022-05-16:
We are working on new compiler toolchain. 
Meanwhile you can downgrade to 10.3 version, it should work fine.

itizir, 2022-06-10:
The problem is not exclusive to M1: basically same thing on Intel chip.
Thanks for the detailed workaround instructions!

J3G0, 2022-07-08:
Version 10.3 of GCC works

skotopes, 2022-07-08:
We almost finished fix. Will be included in next release

hedger, 2022-07-20:
@adamm please try building latest dev. It should download toolchain & everything necessary.
Reopen if you encounter any issues.

closed 2022-07-20 by hedger