flipperdevices/flipperzero-firmware #1234: Two-dimensional layout for saved remotes

lykahb, 2022-05-15:
**Is your feature request related to a problem? Please describe.**
The buttons on a saved remote are displayed as a list. It may take long time to scroll to the needed button.

**Describe the solution you'd like**
The universal library for TV has a scene with a two-dimensional layout. A saved remote would display a similar layout for the names that it recognizes, and puts the rest in a list below. For example, a saved remote has these commands: POWER, VOL+, VOL-, CH+, CH-, NETFLIX, YOUTUBE. Then we display the first commands as icons in a layout, and put the netflix and youtube in a list below, 

**Describe alternatives you've considered**
A simpler grid with two columns that have names and no icons. The items are displayed in the same order as in the .ir file, with no mapping to icon or layout position. The names are truncated to fit half width of the screen.

**Additional context**
Add any other context or screenshots about the feature request here.

skotopes, 2022-05-16:
Hi, we have it backlog, most likely it will be implemented after we finish with Infrared refactoring (summer)

Summertime, 2022-05-30:
3 columns IMO would be perfect, and like a small set of icons

```
name: Power
icon: POWER
width: 1

name: Source
width: 2

name: 7
width: 1

name: 8
width: 1

name: 9
width: 1

[etc etc]

# unselectable, does not render as button,
# may have a name still to show text
# e.g. a "V" between vol up and vol down
blank: 1
width: 1

name: 0
width: 1

blank: 1
width: 1
```
```
[ P ][ Source ]
[ 7 ][ 8 ][ 9 ]
[ 6 ][ 5 ][ 4 ]
[ 3 ][ 2 ][ 1 ]
     [ 0 ]
```

tbh I can't think of any other option other than 3 cols, just because of numpads

Light-Wing, 2022-05-30:
I think the source button should have its own icon as well, as its a button that should exist on every remote...

And maybe making sub menus possible can help with a menu control sun menu (a button to get into the menu and arrow keys to navigate), and a sub menu for direct channel number input,  and other popular buttons a remote can have...

hasmar04, 2022-06-01:
Some of this stuff could be made easier to configure through a wizard. For instance, it tells you to press power and saves that as the power button etc. Easier than recognizing names for common buttons.

themaks, 2022-07-18:
> I think the source button should have its own icon as well, as its a button that should exist on every remote...
>
> And maybe making sub menus possible can help with a menu control sun menu (a button [...]

Completely agree. One submenu could even directly map the arrow wheel and OK button of the Flipper to arrows (rotated by 90°) and OK button of the remote, and the "back" button would exit this submenu.

Light-Wing, 2022-10-11:
I've seen a few apps that use long press back button to exit, and short
press as a extra button (for example, back press of remote)

ikoas, 2023-09-12:
I'm a bit late but here's my idea for a simplified UI for IR remotes.

- Using the base buttons as shortcuts as it follows:
    - Up
    - Down
    - Left
    - Right
    - Select
    - Long Up
    - Long Down
    - Long Left
    - Long Right
    - Long Select
- Map saved IR signals to the previous shortcuts.
- Being able to map saved IR signals from different remotes.
- Add "submenus". You can map one of the previous shortcuts to open another map of shortcuts. (Something akin to a radial menu)
- Being able to select between list and simplified UI for each menu/submenu.
- Use the back button to go back through the "submenus"
- Show a list of the current "submenu" path (Living room > TV > Directions )
- Use a long back press to exit the IR app or go back to the main IP app menu.