flipperdevices/flipperzero-firmware #1239: Disk Access Error

AFstache, 2022-05-16:
I tried updating firmware a couple of weeks ago and qFlipper couldn't see my device, so I put it in DFU mode and was able to install the firmware, but I had lost all of my level up progress. I went to restore the backup that I had made a day or two prior, I couldn't upload the backup. When I use qFlipper, I get a "Failed to process backup directory" and when I try and use the ttyACM0 serial port through minicom, even trying to create a backup I get Result:FAIL. I have since factory reset my device, hoping that might clear things up, but it hasn't.

skotopes, 2022-05-20:
Hi.

Can you provide logs from companion and flipper?

AFstache, 2022-05-20:
Logs from terminal

14844 [DEV] Restore Internal storage @R3gan3r START
14846 [RPC] (2) Gui Start VirtualDisplay START
14856 [UTL] Restore /int @R3gan3r START
14883 [DBG] Requested directory not found
14883 [UTL] Restore /int @R3gan3r ERROR: Failed to process backup directory
14884 [DEV] Restore Internal storage @R3gan3r ERROR: Failed to process backup directory
14885 [BKD] Current operation finished with error: "Failed to process backup directory"
14914 [RPC] (2) Gui Start VirtualDisplay SUCCESS

skotopes, 2022-05-20:
that may happen if your backup is too big for internal storage, how big it is?

AFstache, 2022-05-20:
20k. I don't understand how it could be too big. I even tried again after a factory restore. I downloaded a clean backup and tried to turn around and push it back and got the same error

skotopes, 2022-05-20:
Do you use SD card?

AFstache, 2022-05-20:
Yes I do

skotopes, 2022-05-20:
then your internal storage is abnormal. Can you show file listing?

AFstache, 2022-05-25:
>: storage list /int                                                                                                    
        [F] bt.settings 9b                                                                                              
        [F] desktop.settings 28b                                                                                        
        [F] dolphin.state 40b                                                                                           
        [F] notification.settings 24b

simsnet, 2022-06-15:
I have the same issue, running qFlipper 1.1.0.  After examining the [user restore code](https://github.com/flipperdevices/qFlipper/blob/d550f8e2b605ce76abaab7cdbae63329da820635/backend/flipperzero/utility/userrestoreoperation.cpp#L61-L65), I saw how qFlipper looks for backup files.

Right now to restore a backup, you must select the folder that the **_backup folder_** is in.  For instance, if your backup is saved on the desktop as `YourFlipperName/int/...`, you'll need to select the desktop folder in order to restore.  qFlipper will then locate a folder under your Flipper name and restore the files.

Put the backup folder in a folder of its own, like `/NewFolder/YourFlipperName/int/...`, then in qFlipper click "Restore" and select `NewFolder`.

skotopes, 2022-07-10:
Please check latest dev and reopen if issue is still there.

closed 2022-07-10 by skotopes