flipperdevices/flipperzero-firmware #1240: Expose a USB-GPIO bridge

noneucat, 2022-05-16:
**Is your feature request related to a problem? Please describe.**
It would be useful for the Flipper Zero to have a mode that exposes GPIO pins over USB. 

**Describe the solution you'd like**
If the Zero is able to emulate a USB interface for a device that already exposes GPIO (like an [FT245R](https://electronics.stackexchange.com/a/61301)), this would allow PCs to interface with the Zero with using existing GPIO libraries.

**Describe alternatives you've considered**
Currently kludging a project using the USB-UART interface, but it would be nice to have access to the GPIOs directly. 

**Additional context**
n/a

BJamin99, 2022-05-17:
Might be possible to do this by exposing it through an additional USB interface along side the USB-to-UART?  Might be interesting to also add support for this interface as an additional board to the Adafruit Blinka project similar to the FT232/FT2232/MCP2221/GreatFET One.

SamuelYvon, 2022-05-24:
I'm also looking for this feature and working on implementing it. There already is a start of GPIO feature through the CLI bridge provided by the flipper; I was thinking of extending it to allow binary read/write. A small client interface could be written in python. I also saw a beginning of a RPC interface; maybe it would be possible to control the GPIO pins through that but I haven't looked into it.

BJamin99, 2022-05-24:
Also saw that there may be work on a BusPirate Mode too.

noneucat, 2022-05-26:
> I'm also looking for this feature and working on implementing it. There already is a start of GPIO feature through the CLI bridge provided by the flipper; I was thinking of extending it to allow [...]

Fantastic @SamuelYvon! Looking forward to it :)

aprosvetova, 2022-05-27:
> I'm also looking for this feature and working on implementing it. There already is a start of GPIO feature through the CLI bridge provided by the flipper; I was thinking of extending it to allow [...]

These enhancements might be useful for humans interacting with CLI, but I strongly advise against using CLI in automated software.

CLI is meant to be used by people and we're trying to keep it as user-friendly as possible.

RPC is the perfect way to go if you want to control your Flipper with software.
Maybe our protobuf schema needs a GPIO subsystem?..

SamuelYvon, 2022-05-27:
> These enhancements might be useful for humans interacting with CLI, but I strongly advise against using CLI in automated software.

I agree, my principal interest right now is GPIO through CLI (for humans). If someone can guide me towards the protobuf format and how I could add such a subsys. I'd gladly make a PR later

SamuelYvon, 2022-05-28:
Ok; I figured out the whole RPC thing, so I'm going to implement that as well. I'll link the PR when it's done / in a draft-worthy state

SamuelYvon, 2022-05-28:
> Maybe our protobuf schema needs a GPIO subsystem?..
See https://github.com/flipperdevices/flipperzero-protobuf/pull/30

I added python bindings in https://github.com/flipperdevices/flipperzero_protobuf_py/pull/5 and the PR for the firmware repo is linked above :)

skotopes, 2022-07-10:
PRs by @SamuelYvon are merged. Please check.

closed 2022-07-10 by skotopes

mjsir911, 2024-02-13:
Did this end up exposing a usb gpio device, or is it only accessible through the cli?

This doesn't really solve the issue of interfacing with the flipper's gpio pins through standard gpio libraries.