flipperdevices/flipperzero-firmware #1247: Frequency Analyzer

RogueMaster, 2022-05-19:
FA no longer works as of the latest Megacode/FA changes. It's either not working or not sensitive enough. I am testing with the remote directly next to the back of the flipper.

Skorpionm, 2022-05-20:
you don't need to bring it to the back of the flipper, you need to bring it from the left side, the antenna is there.  and what frequency are you analyzing?  if this is a car alarm, then the transmission there is very short, you need to press the button several times

Skorpionm, 2022-05-23:
https://github.com/flipperdevices/flipperzero-firmware/pull/1262

corrected the work of the analyzer, it should work better, but the remote control still needs to be leaned on the left side

RogueMaster, 2022-05-23:
I've tried on all sides. Detection is very fast before the #1221, after those changes, detection does not work at all for the remote. No matter what side it is on.

RogueMaster, 2022-05-23:
I will try with #1221 and #1262 once its in dev

closed 2022-05-24 by RogueMaster