flipperdevices/flipperzero-firmware #1249: Local sub Ghz Chat (Whistle)

OnyxtheDragon, 2022-05-19:
**Describe the solution you'd like**
A pictochat style local communication application named "Whistle" utilizing the sub Ghz radio and the mobile companion app.
(Obviously name isnt set in stone i just figured it sounded neat)

Flipper would function as the communication radio with the mobile app presenting a keyboard and discord style interface. 

Communicaiton could be divided into "pods" similar to channels within a server

Users would have a nickname and an ID. The nickname would be customizable and is set when joining a "Pod", IDs use the name of the flipper and function as a UUID.

[image]
(rough layout of message composition, featuring: nickname, time sent, and signal strength)

[image]
(Rough layout of additional information screen, Opened by tapping on someones nickname)

**Additional context**
While obviously not a necessary feature, it would be quite interesting to use at a convention or other large gatherings where multiple flippers are present.

11OUTsane11, 2022-05-19:
Would be cool. Especially if useable, in some capacity, without the app too.

skotopes, 2022-05-20:
Hi, as I told in discord we already have this under `subghz chat` cli command.
At this point we don't plan to implement UI for it. We will return to this topic after releasing v 1.0.0.

closed 2022-05-20 by skotopes