flipperdevices/flipperzero-firmware #1252: Sub-GHz Read RAW function does not completely erase when tapping Erase

pliable, 2022-05-20:
**Describe the bug**
When using Read RAW under the Sub-GHz module, after it's done recording, tapping left will not erase the recording completely (it only seems to clear the screen). The raw recording will still appear under Saved

**To Reproduce**
1. Press OK button
2. Sub-GHz -> Read RAW
3. Press OK button to record
4. Press OK button to stop recording
5. Press Left to delete
6. (screen clears at this point)
7. Go back one menu
8. Select Saved
9. Raw_signal_ file will be present

**Expected behavior**
Seems like it should erase the RAW recording completely, not just clear it from the screen? At least that's what I gather it should do

**Additional context**
It make me sad :(

Skorpionm, 2022-05-20:
no, it shouldn't.  when recording a RAW file, the file is immediately written to SD, since the flipper does not have enough RAM to store such large files, with a new recording, the old file will be overwritten and a new one will be written or renamed when saving.  but if suddenly you pressed the wrong button, you still have a chance to save the record by renaming the temp file manually

pliable, 2022-05-20:
And here I got all excited cause I thought I was helping :(. Perhaps, the wording needs to be changed to be more clear? I know I wasn't the only one who thought this was a bug.

Anyway, thank you for all your work 🙏🏼🙏🏼🙏🏼. Uh, I'll let you close it, I'll keep it open just in case lol

Skorpionm, 2022-05-22:
https://github.com/flipperdevices/flipperzero-firmware/pull/1262
fix

closed 2022-06-02 by Skorpionm