flipperdevices/flipperzero-firmware #1263: [subghz] "Add Manually" menu missing Security+ 1.0/2.0 protocols

digitalcircuit, 2022-05-23:
**Is your feature request related to a problem? Please describe.**
In the SubGhz app, the `Add Manually` menu doesn't include the option to create new/random Security+ 1.0 or 2.0 remotes.

[According to what's been shared on the Flipper forums](https://forum.flipperzero.one/t/request-to-support-chamberlain-rolling-code/3157/13 ), this should not be a legal issue since it's creating a unique remote to train with your garage door opener, much like buying a new replacement remote from a store.

**Describe the solution you'd like**
The SubGhz app gains support for creating random Security+ remotes [alongside existing dynamic code generation](https://github.com/flipperdevices/flipperzero-firmware/blob/f04d0eea9683d45cea6b8761f4598fcbf07e023d/applications/subghz/scenes/subghz_scene_set_type.c#L11-L24 ).

**Describe alternatives you've considered**
* Creating a new static (non-dynamic) remote code and teaching that to the garage door
  * Static codes are a security risk

**Additional context**
Access the menu via Flipper desktop → `OK` button → `Sub-GHz` → `Add Manually`.

The relevant patents as mentioned in [a Security+ reverse-engineering repository](https://github.com/argilo/secplus ) are either [expired for Security+ 1.0](https://patents.google.com/patent/US6980655B2/ ) or [abandoned for Security+ 2.0](https://patents.google.com/patent/US20110317835A1/ ), so creating *new/unique* remotes (again, **not** cloning existing remotes) seems unlikely to be an issue.  I am not a lawyer though.

Skorpionm, 2022-05-23:
if the functionality for creating and transferring such codes is implemented, then this will not stop people at all from clanging and not legally using them

digitalcircuit, 2022-05-23:
@Skorpionm Good point - I overlooked that the new randomized remote needs stored somewhere.  When briefly checking the code, I also mixed up the `CAME Twee` protocol (static, allowed) and `CAME Atomo` (dynamic, not allowed).

Feel free to close this issue if desired.

While I respectfully disagree with this reasoning on a technical level, I appreciate that there's significant concern and caution around ensuring the Flipper remains legal, especially this early on (preorder stage).  Not all lawyers would have the technical insight to recognize intent if misunderstandings arose, and this is assuming the Flipper team would even get a chance to defend themselves.

It might be worthwhile clarifying this within the Flipper team to help avoid confusion ([as per this forum post referenced previously]( https://forum.flipperzero.one/t/request-to-support-chamberlain-rolling-code/3157/13 )).  I apologize for dragging this issue back up - I'll keep this in mind!

closed 2022-05-23 by digitalcircuit

Skorpionm, 2022-05-24:
https://github.com/flipperdevices/flipperzero-firmware/pull/1268
Security+ 1.0

Skorpionm, 2022-05-26:
https://github.com/flipperdevices/flipperzero-firmware/pull/1273
Security+2.0