flipperdevices/flipperzero-firmware #1274: realloc implementation memcpy's from source with invalid size

factor27, 2022-05-27:
See https://github.com/flipperdevices/flipperzero-firmware/blob/dev/core/furi/memmgr.c 

```
void* realloc(void* ptr, size_t size) {
    if(size == 0) {
        vPortFree(ptr);
        return NULL;
    }

    void* p = pvPortMalloc(size);
    if(ptr != NULL) {
        memcpy(p, ptr, size);
        vPortFree(ptr);
    }

    return p;
}
```

The memcpy is based on the new size.  This is incorrect, the spec says to copy the min of size allocated for the passed in ptr and new size.  If the old size was smaller, the remaining pages may not be mapped (or may leak data from another alloc).

skotopes, 2022-07-11:
Hi. That was made intentionally.

closed 2022-07-11 by skotopes

factor27, 2022-07-11:
I don't understand.  How do you ensure the ptr is large enough to copy the new size?

You haven't addressed the points I raised or now this implementation does not align with the std realloc.  Are you saying you're intentionally copying too much data (including unmapped or data from other allocs)?

There are also no comments stating any assumptions, which it sounds like you're relying on.

skotopes, 2022-07-11:
Let me try to explain:

- Basically we have RAM1 region, where we have heap
- This is physical memory and it is safe to read from any address in it
- So when we copy memory overread is ok
- Also our memory allocator doesn't know allocated block size and calculation it is not always possible
- Extra space will be filled with garbage, but this is up to implementation, so it's ok

PS: I'll add comments to the code

factor27, 2022-07-11:
Thanks for the explanation - I didn't know this code operated on physical addresses.

Comments would be appreciated!