flipperdevices/flipperzero-firmware #1276: Power off on return long press

azaitsev, 2022-05-27:
**Is your feature request related to a problem? Please describe.**
Sometimes I need to quickly power off the device.

**Describe the solution you'd like**
It would be nice to use long press on return button to power off device. Could you implement behaviour like on old mobile phones or mp3 players? I see long press on return is not used on main screen.

Light-Wing, 2022-05-27:
I'm quite sure if you press left button and back button it shuts down...

this also might help
https://docs.flipperzero.one/basics/power#og-flipper-power-modes

azaitsev, 2022-05-27:
Transport mode — in this mode the device is supplied to users. Only the built-in RTC domain clock is powered. 
The transport mode is activated through the menu: Settings -> Power -> Power OFF.

I'd like to get this just by long press of return.

closed 2022-06-10 by skotopes