flipperdevices/flipperzero-firmware #1278: Sub-GHz send from "Read" causes Hard Fault

camercu, 2022-05-27:
**Describe the bug**
After recording keyfob using Sub-GHz Read, clicking "Send" causes firmware Hard Fault, requiring repair with qFlipper app to use the Flipper again.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on...
2. Press center (menu) button
3. Select 'Sub-GHz'
4. Select 'Read'
5. Select 'Config'
6. Set Frequency to 433.92, Hopping OFF, Modulation AM270
7. Record Subaru Forester keyfob lock signal
8. Select captured signal
9. Select 'Send'
10. Flipper crashes, screen blank, buttons unresponsive (requires repair with qFlipper)

**Expected behavior**
The Flipper should gracefully replay the recorded signal.

**Logs**
Not sure where to find these... If you provide instructions on how to grab them, I can provide logs.

**Target**
Flipper Zero Model FZ.1. FCC ID 2A2V6-FZ, IC: 27624-FZ.

HW Version: 12.F7B9C6 R02 Aer0m0

Serial Number: C7526D0126E18000

FW Version: 0.58.1 (12 May 2022), cb6cf957(1260)

**Additional context**
Example recorded data from 2010 Subaru Forester keyfob (rolling code, so changes every time):

```
CAME TWEE 92bit433.92AM
Key:0xFF77E3D451040048
Btn:B
DIP:0000011110
```

Recorded on 433.92 AM270.
FCC ID of keyfob: [NHVWB1U711](https://fccid.io/NHVWB1U711)

DrZlo13, 2022-05-27:
> Flipper crashes, screen blank, buttons unresponsive (requires repair with qFlipper)
Even LEFT+BACK doesn't work?
https://docs.flipperzero.one/basics/control#ih-reboot

Skorpionm, 2022-05-27:
so let's start with the fact that who said that SUBARU is a supported protocol, secondly who created such a SUB file?

Skorpionm, 2022-05-27:
and what does not work is expected, and should not. the flipper did not correctly accept the protocol the fastest. but I can’t fix it until I see it at least in RAW recording. in ideal you need a remote control from Subaru

Skorpionm, 2022-05-27:
subaru should look like this
Packet: 55 77 17 F1 17 81 81 85 03 DE 
-----------------------------------------
CRC: 0E
Calculated CRC: 0E
Command: lock
rollingcode = 8503D
next rollingcode = 8503E
Next packet (unlock): 55 77 17 F1 17 82 82 85 03 EF 
Next packet (lock): 55 77 17 F1 17 81 81 85 03 EF

camercu, 2022-06-03:
> > Flipper crashes, screen blank, buttons unresponsive (requires repair with qFlipper)
> 
> Even LEFT+BACK doesn't work? https://docs.flipperzero.one/basics/control#ih-reboot

LEFT+BACK did work, thank you. I wasn't aware of that feature.

camercu, 2022-06-03:
> Can you provide us .sub file?

Please forgive me, I'm new to hardware and RF (coming from software dev background, was hoping to use the Flipper as a way to get introduced into the hardware/RF worlds). What is a .sub file, and where can I find it?

camercu, 2022-06-03:
> so let's start with the fact that who said that SUBARU is a supported protocol, secondly CAME TWEEN is not like SUBARU

I wasn't sure whether it knew the protocol for subaru or not, so I just tried it to see, and CAME TWEEN was how it was recognized. Either way, I still expected it not to crash when replaying a signal that it had read.

Skorpionm, 2022-06-07:
> > Можете ли вы предоставить нам файл .sub?
> 
> Пожалуйста, простите меня, я новичок в аппаратном обеспечении и радиочастотах (исходя из опыта разработки программного обеспечения, я надеялся использовать Flipper как способ познакомиться с миром аппаратного / радиочастотного оборудования).~Что такое файл .sub и где его найти?~
> 
> Вот файл .sub из выполнения "чтения необработанных данных" на Subaru за несколько нажатий кнопки блокировки.
> 
> [Subaru_forester2010.sub.zip](https://github.com/flipperdevices/flipperzero-firmware/files/8829166/Subaru_forester2010.sub.zip)
thanks for raw, I'll see if there is enough data to add support for Subaru, but without opening other people's cars

Skorpionm, 2022-06-07:
[image]
Manchester encoding, symbol rate 1000
data in your record
0x8881C2BAEFBFC76
0x8881C2BAEFBFC65
0x8881C2BAEFBFC54
0x8881C2BAEFBFC43
0x8881C2BAEFBFC32
you can see how the counter is decrementing
need more entries. namely, 5+ clicks on each keyfob button for at least 2 seconds and at least 1 record from 3+ clicks with a button hold time of 15 seconds. and sign everything where, I think I can add this protocol

skotopes, 2022-07-10:
Please check latest dev and reopen if issue is still there.

arty-hlr, 2022-07-27:
Same issue here with latest firmware, just received my Flipper today. Also on CAME TWEEN. Please reopen? Or should I open a separate issue?

Skorpionm, 2022-07-27:
0.63?

camercu, 2022-07-28:
> [image] Manchester encoding, symbol rate 1000 data in your record 0x8881C2BAEFBFC76 0x8881C2BAEFBFC65 0x8881C2BAEFBFC54 0x8881C2BAEFBFC43 0x8881C2BAEFBFC32 you can see how the counter is [...]

I captured a larger sample of clicks from the 2010 Subaru Forester key fob for you to analyze.

[Forester-capture.zip](https://github.com/flipperdevices/flipperzero-firmware/files/9204400/Forester-capture.zip)

For my own learning, what are you using to do the analysis? How can I do that myself? The screenshot you posted looks almost like you loaded the file into the Saleae Logic2 program, but I couldn't figure out how to do that. I'd love to be able to do signal analysis, if you're willing to teach a man to fish!

skotopes, 2022-09-06:
Check latest Release and reopen if issue persist.

closed 2022-09-06 by skotopes

camercu, 2023-01-08:
@Skorpionm how did you get to view the .sub file's data in the Saleae Logic program? I thought I could [write a parser](https://gist.github.com/camercu/e5805cdedf176927b0c64147991121d1) that converts .sub files to the Logic2 [binary export format](https://support.saleae.com/faq/technical-faq/binary-export-format-logic-2) for digital data (.bin files), but then I realized that won't work because you can't open those files in Logic2.