flipperdevices/flipperzero-firmware #1280: Require pin after power on or reboot

azaitsev, 2022-05-27:
**Is your feature request related to a problem? Please describe.**
I'd like to improve security of the device.

**Describe the solution you'd like**
If Pin is set we need ask for Pin after reboot or power on.

digitalcircuit, 2022-05-31:
If you lock your Flipper, then force reboot (holding back + left), it'll be locked when it starts back up.  I think [the team intentionally changed it to only lock on reboot if it was locked](https://github.com/flipperdevices/flipperzero-firmware/pull/1167) to fix [breaking qFlipper's update process](https://github.com/flipperdevices/flipperzero-firmware/issues/1178) (which involves multiple reboots). 

Currently, there's [not any way to automatically lock with PIN after a delay](https://github.com/flipperdevices/flipperzero-firmware/issues/1186), which when implemented might solve your concern.

skotopes, 2022-07-10:
Yep. Current locking scheme is designed to handle special cases. We haven't found better solution yet.

closed 2022-07-10 by skotopes