flipperdevices/flipperzero-firmware #1281: Intellicode 2 sub ghz protocol

Gustavas007, 2022-05-28:
Thanks for getting security+ 2.0 up and running. Can we get something similar for intellicode 2?

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Skorpionm, 2022-05-28:
https://forum.flipperzero.one/t/my-remote-isnt-supported-how-to-add-new-sub-ghz-protocol-in-flipper-zero/2033/6

Gustavas007, 2022-05-28:
Thank you.

https://forum.flipperzero.one/t/intellicode-2-code-dodger-2/3389

skotopes, 2022-07-10:
Please reopen with required data.

closed 2022-07-10 by skotopes