flipperdevices/flipperzero-firmware #1283: Bluetooth debug: "Incorrect Radio Stack"

silascutler, 2022-05-29:
**Describe the bug**
Bluetooth debug does not run and consistently returns a message "Incorrect Radio Stack".  I've tested with both stock firmware and releases. 

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on...
2. Press center (menu) button
3. Select 'Settings'
4. Select 'System'
5. Set 'Debug' to `Enable`
6. Press back button (to settings menu)
7. Press back button (to menu)
8. Select 'Debug Tools'
9. Select 'Bluetooth Debug'
10. ["Incorrect Radio Stack" error message is displayed]

**Expected behavior**
Details of system bluetooth information: hw address, nearby signals, etc

**Logs**
```
576818 [I][LoaderSrv]: Starting: Bluetooth Debug
576824 [I][AnimationManager]: Unload animation 'L1_Furippa1_128x64'
576828 [E][BtDebugApp]: Incorrect radio stack, replace with HciLayer for tests.
```

**Target**
```
device_info_major       : 2
device_info_minor       : 0
hardware_model          : Flipper Zero
hardware_uid            : [REMOVED]
hardware_otp_ver        : 2
hardware_timestamp      : 1649667915
hardware_ver            : 12
hardware_target         : 7
hardware_body           : 9
hardware_connect        : 6
hardware_display        : 1
hardware_color          : 1
hardware_region         : 2
hardware_name           : Rariha
firmware_commit         : cb6cf957
firmware_commit_dirty   : false
firmware_branch         : 0.58.1
firmware_branch_num     : 1260
firmware_version        : 0.58.1
firmware_build_date     : 12-05-2022
firmware_target         : 7
radio_alive             : true
radio_mode              : Stack
radio_fus_major         : 1
radio_fus_minor         : 2
radio_fus_sub           : 0
radio_fus_sram2b        : 16K
radio_fus_sram2a        : 0K
radio_fus_flash         : 24K
radio_stack_type        : 3
radio_stack_major       : 1
radio_stack_minor       : 13
radio_stack_sub         : 3
radio_stack_branch      : 0
radio_stack_release     : 2
radio_stack_sram2b      : 19K
radio_stack_sram2a      : 14K
radio_stack_sram1       : 0K
radio_stack_flash       : 120K
radio_ble_mac           : [REMOVED]
enclave_valid_keys      : 10
enclave_valid           : true
protobuf_version_major  : 0
protobuf_version_minor  : 7
```

**Additional context**
Add any other context about the problem here.

Astrrra, 2022-05-29:
Yes, you need a different radio stack to run bluetooth debug.

silascutler, 2022-05-29:
I saw that mentioned on the forum.  On the standard hw stack, it looks like it runs fine when the `if(furi_hal_bt_get_radio_stack() != FuriHalBtStackHciLayer) {` from [1] is commented out. 

[1] https://github.com/flipperdevices/flipperzero-firmware/blob/dev/applications/bt/bt_debug_app/bt_debug_app.c#L100

skotopes, 2022-05-30:
Yep, those are debug tools that are requiring different stack type, only meant to be used at factory.
Intended and tested to be working with hci_layer stack.

closed 2022-05-30 by skotopes