flipperdevices/flipperzero-firmware #1284: Support for NFC NTAG203

evilpete, 2022-05-31:
Flipper will scan NTAG215 but not the simular/simpler lesser memory NTAG203 tags

NTAG203 and the NTAG215 are from the same NTAG family of chips (Both Type 2) ( ECC signatures are not supported in the NTAG203)

GMMan, 2022-06-15:
I can take this on after #1313 is merged. Note that NTAG203 is quite different from NTAG21x from the lack of configuration area, and is not a simple extension of original Mifare Ultralight due to the in-array counters and odd dynamic lock bytes, so there's some effort required to emulate it properly.

GMMan, 2022-06-30:
Ran into a little snag with this, I'm not sure where to source actual NTAG203s. The cards Adafruit sells are counterfeit NTAG213s.

evilpete, 2022-06-30:
want me to mail you a couple??

I've been looking into it, it seems they're obsolete yet at the same time
they are  still in use everywhere.

GMMan, 2022-06-30:
Currently asking Adafruit to confirm whether they have an inventory of any NTAG203s, but if they don't, then sure.

GMMan, 2022-07-07:
Adafruit said their manufacturer switched chips on them, so their stock is not guaranteed to be NTAG203. I've sent you an email to the address you have on your commits.

evilpete, 2022-07-07:
no problem I'll dig a few out and the other later today,  direct email me
the shipping address.

skotopes, 2022-07-10:
@gornekich what do you think? Can we include it in our plans?

GMMan, 2022-07-21:
@evilpete Got the tags, and finished up the PR.

PR #1383 (2022-07-25): nfc: NTAG203 support
# What's new

Resolves #1284

- NTAG203 read support
- NTAG203 full emulation, including counters
- NTAG203 option in "Add manually" menu

Research notes [here](https://discord.com/channels/740930220399525928/997271017401630852/999538438250450984).

Note: could not test counter lock behavior since I ran out of tags to test counters on, but I presume it works like any of the other locked pages by NAKing if you try to write to it.

# Verification 

- NTAG203 tag can be read and saved.
- Emulated NTAG203 tag works the same as the original and follows behavior outlined by the  [... cut]
closed 2022-07-25 by skotopes