flipperdevices/flipperzero-firmware #1290: Rename LiftMaster_3XX to Security+1.0_3XX

xslim, 2022-06-02:
**Is your feature request related to a problem? Please describe.**
While Reading the Sub-GHz codes from Remote, I see Security+1.0.
But when I try to create a remote Manually, it is not there. 
Digging thru source code, I see that LiftMaster is a Security+1.0.

**Describe the solution you'd like**
I would like to see a Security+1.0 in Add Manual menu, as it is more generic, and not only related to garage opener.

**Describe alternatives you've considered**
Alternative: Add Security+1.0 in addition to LiftMaster, but than we are just adding duplicated entries.

**Additional context**
Add any other context or screenshots about the feature request here.

xslim, 2022-06-02:
Example 3-button remote control that is Security+1.0 + Security+2.0 is Intermatic RC939

hasmar04, 2022-06-05:
As an extension to this, I have a remote which uses the Security+ 1.0 protocol but on the 433.92 AM frequency, so possibly an entry with changeable frequency?