flipperdevices/flipperzero-firmware #1296: Chosen custom sub-GHz frequency gets reset when going from Read/Read RAW to config screen

ide, 2022-06-03:
**Describe the bug**
When the sub-GHz scanner is listening on a custom frequency, going back to the config screen resets the frequency to the nearest regional frequency that is higher than the custom frequency. For example, if the custom frequency is 314.35 MHz and the Flipper's regional frequencies include 315.00 MHz, the Flipper will "round up" 314.35 MHz to 315.00 MHz when going back to the config screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a `subghz/assets/setting_user` file with a custom frequency such as `frequency: 314350000`
2. Run the Sub-GHz app, choose Read, choose Config. On the config screen, choose the custom frequency (314.35 MHz). Currently the custom frequencies are at the end of the list.
3. Press Back to go to the scanning screen and confirm the Flipper is listening for 314.35 MHz.
4. Press Left to go to the config screen. The selected frequency changes to the next highest regional frequency (315.00 MHz) instead of staying on the custom frequency.

Note: this may just be a GUI issue. Even though the config screen shows 315.00 MHz, going back to the scanning screen shows 314.35 MHz.

**Expected behavior**
The selected frequency should not change when navigating between the scanner screen and config screen.

**Logs**
The logs don't look very helpful but I got them by setting the Flipper's log level to "Debug" and reading the logs with an ESP32, which gives much more detailed logs than qFlipper does:

```
11945 [I][LoaderSrv]: Starting: Sub-GHz
11951 [I][AnimationManager]: Unload animation 'L1_Sleep_128x64'
11970 [T][StorageAPI]: File/Dir 0x11928 alloc
11978 [T][StorageAPI]: File 0x11928 - 0x12c94 open (/ext/subghz/assets/setting_user)
11997 [I][SubGhzSetting]: Keeping standard frequencies
12003 [I][SubGhzSetting]: Frequency loaded 314350000
12007 [I][SubGhzSetting]: Frequency loaded 348000335
12011 [I][SubGhzSetting]: Frequency loaded 387000000
12015 [I][SubGhzSetting]: Frequency loaded 433220000
12018 [I][SubGhzSetting]: Frequency loaded 464000000
12021 [I][SubGhzSetting]: Frequency loaded 779000000
12024 [I][SubGhzSetting]: Frequency loaded 928000000
12040 [I][SubGhzSetting]: Hopper frequency loaded 310000000
12043 [I][SubGhzSetting]: Hopper frequency loaded 315000000
12046 [I][SubGhzSetting]: Hopper frequency loaded 433920000
12051 [I][SubGhzSetting]: Hopper frequency loaded 915000000
12066 [T][StorageAPI]: File 0x11928 - 0x12c94 closed
12067 [T][StorageAPI]: File/Dir 0x11928 free
12071 [I][SubGhzProtocoNiceFlorS]: Loading rainbow table from /ext/subghz/assets/nice_flor_s
12073 [I][SubGhzProtocoCameAtomo]: Loading rainbow table from /ext/subghz/assets/came_atomo
12074 [I][SubGhzKeystore]: Loading keystore /ext/subghz/assets/keeloq_mfcodes
12076 [T][StorageAPI]: File/Dir 0x17b78 alloc
12084 [T][StorageAPI]: File 0x17b78 - 0x17ccc open (/ext/subghz/assets/keeloq_mfcodes)
12141 [T][StorageAPI]: File 0x17b78 - 0x17ccc closed
12143 [T][StorageAPI]: File/Dir 0x17b78 free
12145 [I][SubGhzKeystore]: Loading keystore /ext/subghz/assets/keeloq_mfcodes_user
12147 [T][StorageAPI]: File/Dir 0x17b78 alloc
12156 [T][StorageAPI]: File 0x17b78 - 0x17d84 open (/ext/subghz/assets/keeloq_mfcodes_user)
12172 [T][StorageAPI]: File 0x17b78 - 0x17d84 closed
12173 [T][StorageAPI]: File/Dir 0x17b78 free
12449 [D][BtBatterySvc]: Updating power state characteristic
22848 [D][ViewDispatcher]: View changed while key press 0x20011998 -> 0x20007278. Sending key: Back, type: Release, sequence: 0xf to previous view port
29044 [D][ViewDispatcher]: View changed while key press 0x20011998 -> 0x20007278. Sending key: Back, type: Release, sequence: 0x12 to previous view port
```

**Additional context**
Official firmware v0.59.1

Skorpionm, 2022-06-06:
please attach your /ext/subghz/assets/setting_user i will take a look

Skorpionm, 2022-06-06:
Fix https://github.com/flipperdevices/flipperzero-firmware/pull/1300

ide, 2022-06-06:
This is my `setting_user` file if you still need it:
```
Filetype: Flipper SubGhz Setting File
Version: 1

frequency_default: 315000000

frequency: 314350000
frequency: 348000335
frequency: 387000000
frequency: 433220000
frequency: 464000000
frequency: 779000000
frequency: 928000000

hopper_frequency: 310000000
hopper_frequency: 315000000
hopper_frequency: 433920000
hopper_frequency: 915000000
```

skotopes, 2022-06-09:
Fixed in #1300

closed 2022-06-09 by skotopes