flipperdevices/flipperzero-firmware #1299: Missing pulses on TX - SubGhz

alfie65536, 2022-06-05:
**Describe the bug**
Pulse is missing on TX every 16 pulses on sending 30 microseconds signal

**To Reproduce**
Steps to reproduce the behavior:
1. Create a RAW .sub file with alternating signals (Example: RAW_Data: -30 30 -30 30 and so forth)
2. Play from Flipper device (650AM - 433.9254Mhz)

**Expected behavior**
Using SDR, I expected the values to read as 101010 directly. Result turned out that every 16 pulses, Flipper would not send the signal and skip a value, changing the overall packet.

**Logs**
Using 30 values in the .sub (Example: RAW_Data: -30 30 -30 30 and so forth)
[image]
Sample from the stream:
1010101010101010010101010101010110101010101010100101010101010101101

Using 100 values in the .sub (Example: RAW_Data: -100 100 -100 100 and so forth)
[image]
Sample from the stream:
1010101010101010101010101010101010101010101010101010101010101010101

Closer look between the missing pulse shows failure to send:
[image]

**Target**
Unsure. I was able to find a workaround with the 100 microseconds and might use that. Possibly relay in the Flipper docs that if values intended to be sent are too short, that Flipper may not send reliably?

**Additional context**
Attempting to have Flipper send packets to a RFM69HCW device, exciting project. Trying to have them read as the same with the SDR and attempt to finetune things until they read the same. Did a AAAA repeating hex test to see if they can read the same.

pelrun, 2022-06-06:
Just a note that it doesn't look like pulses are being lost, but instead it's a hard real-time failure where the flipper is too busy servicing a different task and so the next edge is delayed.

Skorpionm, 2022-06-06:
failure is unlikely. just like the flipper is very busy. transfer goes through DMA. if I understood everything correctly, then with durations of 100us everything is fine at 30 it fails? I will see what is really happening there, but I have not seen durations smaller than 150us in any protocol

Skorpionm, 2022-06-06:
[image]
30 us 
[image]

Skorpionm, 2022-06-06:
[image]
I can’t capture such a fast signal, I have a pickup of 50Hz. 
[image]
reception to another module, it seems that CC1101 in RAW format does not have time to send such short signals (in fact, the flipper transmits all data without delay, see photo above), but once again I repeat such short intervals for remote controls

skotopes, 2022-07-10:
Please re-test current dev and reopen issue if problem is still there.

closed 2022-07-10 by skotopes