flipperdevices/flipperzero-firmware #1328: Flipper SubGhz RAW File only support integer bit durations

evilpete, 2022-06-17:
**Describe the bug**

While encoding a Raw RF signal  for Insteon (Home Automation) that required 9124 baud   I ran into a problem due to integer bit (level?) durations

> RAW_Data: -110 220 -220 220 -330 ...

 protocol  requires  9124 baud,  this works out to a bit (level?) durations of 109.600 microseconds  
 
 Given: 9124 baud -> 1/9124 -> 0.00010960105 sec -> 109.60105 ms

When I encode using 110 ms durations the signal is not recognized, if I use 109 ms durations the signal is only recognized approximate 25% of the time.

as a work around I ended up encoding at 109 with 1ms padding every three of four bits, honestly this is a bit of a hack and takes tweaking to work.

**To Reproduce**

On request I  can generate a example file to test but will need the 3 byte address target device

**Expected behavior**

The best workaround I can think of is adding an optional line for baud to the SubGhz File description 

while the CC1101 does not support exact baud  rates it can get within  99.998% (according to my math)

Skorpionm, 2022-06-17:
something is wrong with you even with an accuracy of 5% it is - + 5.75us I doubt very much that it's + -1us

evilpete, 2022-07-01:
when the packet is 364 or  896 a  .4 difference adds up .

skotopes, 2022-07-10:
Theoretically we may add ability to configure timer clocking, please check if TIM2 can be configured for your needs?

evilpete, 2022-07-12:
I was able to work around by tracking the timing offset and adding/deleting 1ms from from the bit duration to keep the packet timing in sync

skotopes, 2022-07-12:
@evilpete also check digital signal library, it's designed to do that automatically

evilpete, 2022-08-05:
> @evilpete also check digital signal library, it's designed to do that automatically

i was working on generated SubGhz RAW, thus can't use the internal signal libaray

closed 2022-09-06 by skotopes