flipperdevices/flipperzero-firmware #1329: Where to suggest documentation suggestions? (docs.flipperzero.one)

patricknomad, 2022-06-17:
I'd like to recommend suggestions for https://docs.flipperzero.one/ for new undocumented features.

There used to be https://github.com/flipperdevices/docs but this is now archived.

Is there another place where PRs can be sent for docs?

DrZlo13, 2022-06-20:
We are working on it, hopefully there will be some solution in the coming weeks.

rcprcp, 2022-08-11:
I would like to +1 this request, as i think crowd sourcing documentation from the community would greatly improve the quality (and quantity) of documentation. 

Thank you.

AkechiShiro, 2022-12-02:
I'm waiting for this as well, to help document some more stuff

skotopes, 2023-03-13:
[image]

Please use this buttons to suggest changes.

closed 2023-03-13 by skotopes