flipperdevices/flipperzero-firmware #1345: MIFARE Classic Anti-Copy Detection

quantum-x, 2022-06-27:
**Background**
In France, almost every apartment building uses RFID Access Control based on MIFARE Classic 1K badges.
The keys to decode these badges are well known (and in Flipper's library), and the badges can be easily cloned / emulated.

However, in France many Access Control Manufacturers now include *anti-copy algorithms* in the *badge contents*.

While the implementation differs between manufacturers, the overall technique remains the same.
Each time a badge is scanned on the building, a _counter_ in the badge contents is updated. 
The next time the badge is scanned, the Access Control Reader _checks the contents_ of the counter.
If the counter is not at the expected value, the badge is rejected.

**The problem**
When badges with anti-copy are cloned - the contents between original badge and the new clone (or emulation) become desynchronised, meaning that the copy will work, but the original will _no longer work_.

If the badge was _emulated_ the copy will no longer work, and the emulation will no longer work either (because its contents weren't updated during the emulation).

Depending on the brand of the badge, typically only one badge is de-activated. However some brands (such as Immotec) will deactivate _all badges associated with an appartment_ when cloned badges are detected.

Although this problem is concentrated in France, it is also present in Belgium, UK, Germany and Australia.

As more and more people in Europe will be getting their hands on Flippers, I feel it's important to discuss the issue, to prevent people unintentionally breaking their residential badges / locking themselves out..

**Suggestions for a solution**
The anti-copy algorithms for each brand and model have a unique fingerprint, and can be detected.
I am unsure the best way to implement (if at all) some type of detection / protection.

- It could be an additional script that can be run against a saved Mifare Classic dump that checks for Anti-Copy
- It could be implemented into the Emulation function, to prevent (or at least warn) of anti-copy presence
- It could be implemented into the Write function (not yet implemented, but on the to-do list) to prevent (or at least warn) of anti-copy presence.

I have the full list of signatures for all brands and models with anti-copy for all of Europe, and will happily contribute this information.

quantum-x, 2022-06-27:
@gornekich Let me know your thoughts on how to implement for the entire community, and next steps so I can provide the anti-copy detection rules.

gornekich, 2022-06-27:
Hello @quantum-x . Thanks for information!

For now we ignore counter related commands during emulation. I can see that this can cause problems, and we try to add working with counters soon. I guess it will solve the issue, when user just copy the card to flipper and no longer use the original card.

Further we will add users possibility to synchronize both variants of card. It will come after we add write functionality. Flipper can detect if the data was updated during emulation, and I think it's a good point to notify user about it. I will talk with our UX designers and come back with solution.

I'm not sure I understood you correctly about **additional script that can be run against a saved Mifare Classic dump that checks for Anti-Copy** Could you explain the solution in more details?

quantum-x, 2022-06-28:
Hi @gornekich 
Thanks for the reply.

To be clear, the anti-copy algorithms in are not simply consecutive counters (0x01, 0x02, 0x03 etc), it's heavily obfuscated data that is written 'randomly' over the badge.

Regarding 'Additional Script that can be run against a Mifare Classic Dump':
 - When a user saves a dump to the Flipper, in the 'Actions' menu (Currently has "Emulate", "Edit UID and Name", "Delete", "Info", etc - it could be interesting to add a "Detect Anti-Copy" function.
 - Users could run this script against a dump to check if the badge has anti-copy signatures
 - This script could also be automatically run before a user emulates or writes a dump, to prevent (or warn) users that the badge contains anti-copy.

In France, ~20% of badges on the market have anti-copy.
Currently if I emulate a badge with anti-copy, **both the original badge and the emulation will be de-activated.**

As EU customers are starting to receive their Flipper devices, there is a very high risk that customers will de-activate their building badges by mistake, with no warning, and no way to repair the badge except to buy a new one (~50EU - 100EU) from the building manager.

Detection is reliable and "easy" to implement, perhaps we should try to implement this quickly.

gornekich, 2022-06-28:
Hello @quantum-x 

We discussed with our team this issue and our priority is to support all commands during Classic emulation.

Regarding anti-copy algorithms, we need at least badges dumps. It would be nice if you send a few dumps with anti-copy signatures and provide an algorithm to detect the signature.

Further, it would be perfect to get these badges and may be readers on hands for tests. Is it possible for you to send them to us?

quantum-x, 2022-06-28:
Hi @gornekich
Thanks for the internal discussion. I understand your team's priorities - and if Mifare Classic Emulation and writing can't be integrated, then integrating anti-copy detection is moot.

Regarding detection and rules, I have full rule-sets for detecting all types of known anti-copy on the market.
It should be mentioned that this comes from the development team at _Parklink Development Limited_ (a company for which I am director). 

We are happy to open-source this code, as it's important for the community. If possible, we'd some acknowledgement for the hard work of the dev team.
Contributors are myself (@quantum-x) @danielebruneo and @triplesprawl 

There are five brands using anti-copy:

- Intratone
- Hexact
- Comelit
- Noralsy
- Urmet

Each brand has different generations of algorithms. 

The `match` function ingests an ASCII HEX representation of the binary dump of a badge.
Code is as follows : 

#### Common method
```
        function hex_substring($hexDump,$startByte,$endByte){
            if($endByte<$startByte){
                return "";
            }
            return substr($hexDump,$startByte*2,($endByte-$startByte)*2+2);
        }
``` 

### Intratone
```
        public function match() {
            return 
                $this->hex_substring($this->hex_dump,0x3c0,0x3cf) == "684578616374202d20434f47454c4543" && //Contains "hExact - COGELEC"
                $this->hex_substring($this->hex_dump,0x3bf,0x3bf) != "00" &&  //Previous byte (part of Sector 14 Key B) not being 00
                $this->hex_substring($this->hex_dump,0x3bf,0x3bf) != "ff" &&  // nor ff
                $this->hex_substring($this->hex_dump,0x3e0,0x3e3) != "00000000" && //IID is present
                $this->hex_substring($this->hex_dump,0x2c2,0x2c2) != "00" &&  // counters not being empty
                $this->hex_substring($this->hex_dump,0x2da,0x2da) != "00" &&  // counters not being empty
                $this->hex_substring($this->hex_dump,0x020,0x02f) != "00000000000000000000000000000000" &&  // block 2 not being 0 filled
                // $this->hex_substring($this->hex_dump,0x380,0x38f) == "00000000000000000000000000000000" &&  // sector 14 being 0 filled (this can be removed)
                $this->hex_substring($this->hex_dump,0x390,0x39f) == "00000000000000000000000000000000" &&  // sector 14 being 0 filled
                $this->hex_substring($this->hex_dump,0x3a0,0x3af) == "00000000000000000000000000000000"     // sector 14 being 0 filled
            ;
        }
```

### Hexact
#### Hexact A
```
        public function match() {
            $hexact_pattern_main = "/4845584143547877880022729a9bd40f484558414354/";
            return 
                preg_match($hexact_pattern_main, $this->hex_dump) &&
                (
                    (
                        $this->hex_substring($this->hex_dump,0x02c0,0x2cf) == "00000000000000000000000000000000" && // 2c* rolling counter block empty
                        $this->hex_substring($this->hex_dump,0x02d0,0x2df) == "00000000000000000000000000000000" && // 2d* rolling counter block empty
                        $this->hex_substring($this->hex_dump,0x02e0,0x2ef) == "00000000000000000000000000000000"    // 2e* rolling counter block empty
                    ) ||
                    (
                        $this->hex_substring($this->hex_dump,0x02c0,0x2cf) == "ffffffffffffffffffffffffffffffff" && // 2c* rolling counter block empty
                        $this->hex_substring($this->hex_dump,0x02d0,0x2df) == "ffffffffffffffffffffffffffffffff" && // 2d* rolling counter block empty
                        $this->hex_substring($this->hex_dump,0x02e0,0x2ef) == "ffffffffffffffffffffffffffffffff"    // 2e* rolling counter block empty
                    )
                )
                || true
[... 66 lines cut ...]
        }
```

### Noralsy
#### Noralsy A
```
        public function match() {
            $noralsy_pattern = "/675a32413770[0-9a-f]{8}395244733978/";
            return preg_match($noralsy_pattern, $this->hex_dump);
        }
```

### Urmet
#### Urmet A
```
        public function match() {
            $urmet_pattern_main = "/8829da9daf767f0788008829da9daf76/";
            return preg_match($urmet_pattern_main, $this->hex_dump);
        }
```

Please note, there are some badges that match A/C rules that do not have anti-copy enabled, but this false positive rate is ~2% maximum.

Hopefully these methods should remove some workload from your dev-team.
Please let me know if I can provide any further assistance.

Foul, 2022-08-16:
got one dump made some times ago from a Cogelec/Intratone badge with all keys.. could help ?
[Intratone.zip](https://github.com/flipperdevices/flipperzero-firmware/files/9353172/Intratone.zip)

quantum-x, 2022-08-16:
> got one dump made some times ago from a Cogelec/Intratone badge with all keys.. could help ? [Intratone.zip](https://github.com/flipperdevices/flipperzero-firmware/files/9353172/Intratone.zip)

This one should indeed trigger the A/C Intratone ("COGELEC") algo.

LoganMcClay, 2022-08-18:
Hello @quantum-x, to see if a badge is really protected, the data should change if dumped before then after a badge is used to enter ? I have a spare Urmet badge with the pattern matching, but the content doesn't seems to be updated when used.

quantum-x, 2022-08-18:
@LoganMcClay The content doesn't necessarily update on each scan.
Depending on the brand, either the badge contents contains the counter, or the access control system contains the counter.

With that said there is a 'before and after' difference between badges that don't have anti-copy enabled, and badges that do have anti-copy enabled - this is what the above algorithms detect.

LilianHori, 2022-08-27:
Ok so my badge is from immotec and first thing I did when I received my flipper was trying to copy it (obviously I didn't read this thread before). It seems it didn't work as I was unable to open my door with my flipper, I will go and test if my badge still work. Should I just stop everything for now about it, should I delete the saved nfc ? This sure is a huge problem and need to be told to EU users

Foul, 2022-08-28:
how does it work when there are several users/badges for the same building to synchronize everything?

quantum-x, 2022-08-28:
If your badge contents matches the IMMOTEC anti-copy rules above, do *not*
attempt to emulate the badge.

IMMOTEC anti-copy will block the copy and the original badge. Depending on
the configuration of the system, it can also erase the badges assigned to
your apartment.

If your badge does not match the anti-copy rules, you should be OK to
emulate it.

Foul, 2022-09-02:
another user posted : 

> I know this remote, and talked to the vendor of it.
> It is not transmitting a usual signal like other remotes do.
> This one sends only 3 little bursts of the signal, like Horman BiSecur, which are hard to catch.
> And it is AES encoded.
> There will be no way to handle this remote with Flipper or any other cloning device or solution!

Yomi2023, 2023-03-28:
HI here,
If somebody it's interested, i have two badge intratone with anty-copy , i'll can make a completly dump
And I have an old card of a packet distributor who is desactived but she had a permanent acces during a certains day before. I dumped it too.
I'm curious to know more about that card and how work the badge of emergency service.
I'll can make test with my two badge and try to modify card and test it.

Foul, 2023-03-28:
The one with permanent access could be interesting ;)

Yomi2023, 2023-03-28:
I had it for four or five days only, now it was desacttived
But I have it and I dumped it

`+Sector: 0
D44BB0715E880400C838002000000020
44001049164916491649000000000000
00000000000000000000000000000000
A0A1A2A3A4A5696789C1EF61A3D48E2A
+Sector: 1
12000000000000000000000000000000
AA07000023020800007A050306004883
5802488358024883017A050308340000
314B4947495679678800EF61A3D48E2A
+Sector: 2
08FFE4C01E2592FAB92020D8FA4CDA1F
2C59BADC37F961267F95A8D57E37A371
20B5B157E6D1380D584304D019D8BE95
314B4947495678778800EF61A3D48E2A
+Sector: 3
18BF51EB4C02B5BFB0BFD40E725B0230
1038959E1861EC68F4D9D93CBBA8A648
78A7BF154A65EB4718C6DC98E9395A63
314B4947495678778800EF61A3D48E2A
+Sector: 4
3844B65951BC756FB21C944D8CE2FDD5
095DB88CE9DE523BE7F983D0B774BE35
00000000000000000000000000000000
314B4947495678778800EF61A3D48E2A
+Sector: 5
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
314B4947495678778800EF61A3D48E2A
+Sector: 6
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
314B4947495678778800EF61A3D48E2A
+Sector: 7
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
314B4947495678778800EF61A3D48E2A
+Sector: 8
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
314B4947495678778800EF61A3D48E2A
+Sector: 9
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
314B4947495678778800EF61A3D48E2A
+Sector: 10
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
314B4947495678778800EF61A3D48E2A
+Sector: 11
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
314B4947495678778800EF61A3D48E2A
+Sector: 12
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
314B4947495678778800EF61A3D48E2A
+Sector: 13
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
314B4947495678778800EF61A3D48E2A
+Sector: 14
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
314B4947495678778800EF61A3D48E2A
+Sector: 15
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
314B4947495678778800EF61A3D48E2A`

Foul, 2023-03-28:
Did you test it successfully ?

Yomi2023, 2023-03-28:
When i found it worked but now she don't work
Do you want the dump?
I search how to post it as a code but it was not posted clearly

Foul, 2023-03-28:
Link to Pastbin.

Yomi2023, 2023-03-28:
Yeah, thanks

[https://pastebin.com/5SbZBcHN](url)

Sweedn, 2023-04-02:
Hello there, any news about this ?
I could have access to some LaPoste cards, but I don't know if this would be usefull.
And yet some Mifare 1k emulation doesnt work, somehting about frequency [FlipperZero forum link with dev answer](https://forum.flipperzero.one/t/vigik-nfc-reader-cannot-read-flippers-emulation/5083/11)

Yomi2023, 2023-04-02:
I could have access to some LaPoste cards, but I don't know if this would be usefull.

Hello,
If you have it physically you'll can dump it with mobile app : mifare claissic tools, and post the dump here

That"s would be great.

Thanl you

quantum-x, 2023-04-02:
> HI here, If somebody it's interested, i have two badge intratone with anty-copy , i'll can make a completly dump And I have an old card of a packet distributor who is desactived but she had a [...]

This is a thread about providing detection of anti-copy algorithms put in place by French Access Control manufacturers to prevent Flipper Zero owners for inadvertently erasing their original cards when they make private copies, or when they make in-scope penetration tests.

@Foul @Yomi2023 @Sweedn You are discussing the VIGIK Service cards, which is out of scope of this thread.
Furthermore, duplication of these cards is illegal and has lead to multiple arrests. I'd suggest removing these posts, or moving the discussion into another forums.

JohnN6, 2023-07-09:
d

chinlung, 2023-08-06:
I want to propose another anti-copy situation: the original card is a Mifare Classic 1K card, SAK:08, KeyA/B are FFFFFFFFFFFFF, so it can be copied and stored by F0, but when F0 simulates, systems like the elevator system and access control system cannot read the content normally (no response at all). Later, I found information on the Internet saying that these systems "may" have a firewall, and need to use a card type called "CUID" (probably the so-called magic card) to work normally. In practice, a copied card using CUID can pass through the above systems normally.

Is it possible for F0 to update the firmware to read data normally through the above systems? Thank you.

jerome-jutteau, 2023-10-20:
Hi!

Thanks for providing the dump and thanks a lot to contributors such as @quantum-x for the hard work.

> got one dump made some times ago from a Cogelec/Intratone badge with all keys.. could help ? [Intratone.zip](https://github.com/flipperdevices/flipperzero-firmware/files/9353172/Intratone.zip)

A first interesting thing to note about this dump is what we see at `0x03c0` -> `0x03cf`:
```
xxd Intratone.dmp
...
000003c0: 6845 7861 6374 202d 0143 4f47 454c 4543  hExact -.COGELEC
...
```

`hExact` is different from `HEXACT` string so this should not match with:
```php
$this->hex_substring($this->hex_dump, 0x03c0, 0x3c5) == "484558414354" && // 3c0-3c5 == HEXACT
```

Hope this help

skotopes, 2024-01-03:
Looks like NFC plugin system solves this issue. Please contribute plugins ;-)

closed 2024-01-03 by skotopes

anonymous10download, 2024-01-04:
What do you mean by the "NFC plugin system", and how does it solve the issue?
 I've been monitoring this issue for an extended period.

skotopes, 2024-01-05:
@anonymous10download NFC plugin system currently covers card data parsing, helps to identify and visualize known systems data. For example: https://github.com/flipperdevices/flipperzero-firmware/pull/3325

quantum-x, 2024-01-15:
As per @skotopes post, Flipper now has plugin / parsing architecture for NFC.
This represents the hardest part of implementing an anti-copy detection plugin.

If someone is available to work on the structure of the plugin, I can contribute up-to-date algorithms for all known brands. Feel free to contact me.

Leptopt1los, 2024-02-03:
> As per @skotopes post, Flipper now has plugin / parsing architecture for NFC. This represents the hardest part of implementing an anti-copy detection plugin.
>
> If someone is available to work on the [...]

@quantum-x contact me via discord (@leptopt1los). I will need a description of the algorithms and dumps for tests

gagzzz, 2024-04-26:
hello,
i'm also interested but rather for the proxmark3. If there was a working anti-copy detection plugin for the Flipper it would probably end up being adapted to work with proxmark3.

Also i wonder if Silca, Copybadge and rebadge do this kind of check as well, i think i've heard copybadge and rebadge sometimes copied protected badges, but i recall a silca copier owner saying it would detect the anticopy straight away. But i also do think Silca is less "intrusive" in the way it clones badges, as it requires an internet connection to work, iirc.

JohnN6, 2024-04-27:
Copybadge.fr claim to be able to re-activate badges which have been disabled when a clone-copy has been used.
This seems to imply that they have direct access to Intratones master database system.    It is uncertain to me if when a copied Vigik is used and then disabled, after they re-enable it, whether you now have **two** working badges or just **one**.

I haven't dared to try.  They say at https://www.copybadge.eu/copybadge/comprendre-les-systemes-anti-copies/

> Les badges de la marque Intratone (groupe Cogelec), sont ceux dont le système anti-copie est le plus efficace.
>La plupart des centrales anti-copies sont connectées à un module GSM qui permet à la marque Intratone d’effectuer des mises à jour à distance pour désactiver les copies et empêcher la reproduction de ses badges. À ce jour, il est impossible de les reproduire.
>Lors d’une copie, le second badge d’immeuble présenté sera refusé par la centrale. Si l’utilisateur teste la copie en premier, elle sera acceptée, néanmoins l’original qui sera présenté ensuite sera considéré comme une copie et sera refusé. Nous avons la possibilité de réactiver à distance un badge Intratone désactivé.

BlackBird63030, 2024-04-27:
Hi, but what about Hexact badges ? I got a dump of my own badge and i want to duplicate it. Any ideas please ?

JohnN6, 2024-04-27:
As to Hexact, I think you will need to dump the badge, use it, dump it again use it again for at least 17 cycles and compare all the copies to see if there are changes recorded on your Hexact fob,.  It seems that some have no anti-copy local on-fob storage used and others have that set up.  IT is a choice made when the building system is set up.

The copy companies suggest that unless co-propietaires ( co-owners ) have agreed and been informed of an exclusivity forbidding copying of entrance badges then they have ( in France ) a right to be able to make and use copies.

Rebadge say they check if copies are allowed, so it might be possible to use them for a test ?
I don't have a Hexact fob so can't help more than this,

BlackBird63030, 2024-04-28:
do you think that the code at the top will be useful ?

BlackBird63030, 2024-04-29:
public function detectAntiCopyFromFile($filePath)
{
    $hexDump = file_get_contents($filePath);

    // Hexact A
    $regexpA = '484558414354787788008FA1D601D0A2(?!0000000000000000000000000000000000000000000000000000000000000000).+?000000000000000000000000000000004845584143547877880089347350BD36';

    // Hexact B
    $regexpB = '^[0-9A-F]{1376}484558414354787788008FA1D601D0A204D26';

    // Hexact C
    $regexpC = '^[0-9A-F]{1376}484558414354787788008FA1D601D0A2^[0-9A-F]{4}6';

    if (preg_match('`' . $regexpA . '`mis', $hexDump)) {
        return "Badge de type Hexact A avec anti-copy";
    } elseif (preg_match('`' . $regexpB . '`mis', $hexDump)) {
        return "Badge de type Hexact B avec anti-copy";
    } elseif (preg_match('`' . $regexpC . '`mis', $hexDump)) {
        return "Badge de type Hexact C avec anti-copy";
    } else {
        return "Badge Hexact sans anti-copy ou type inconnu";
    }
}

$filePath = "chemin/vers/le/fichier/badge_dump.hex";
echo detectAntiCopyFromFile($filePath);

there some code for hexact badges

VincentChalnot, 2024-12-20:
Additional info on this subject for those stumbling upon this topic:

Intratone has 2 different rolling code tech that can be mixed on the same fob.
- The basic one is just a simple counter that is incremented.
  - For my entrance that has an older reader it's incrementing 2 different bytes on sector 11 in 2 different blocks.
  - After realizing that the system had rolling codes, I managed to re-sync the original easily even though the copy had been overwritten by "bruteforcing" a few increments on the proper memory addresses.
- For the more recent readers, it's writing what appears to be random data (is it "truly" random or procedural random or cryptographic, I don't know).
  - The exterior of my building has several readers, all writing 3 bytes each on two different places on sector 11 so 6 random bytes in total.
  - Each reader has it's own address to write those bytes so a fob can be synced on some readers while being desynced on others.
  - Once the fob is desynced, you're screwed especially if you lost the data of the properly synced fob.
  - Two clones can be de-synced on different readers, because each reader has it's own rolling code.

The Flipper Zero is capable of emulating Mifare Classic cards so perfectly that it accepts the rolling code writes during emulation and keeps track of the changes properly. Once a change is written during the emulation process, a new menu item appears on the saved nfc data that allows the user to restore the original data from the initial scan.