flipperdevices/flipperzero-firmware #1353: Support writing NFC tags

virtualdxs, 2022-07-01:
**Is your feature request related to a problem? Please describe.**
I would like to be able to write a new copy of NFC tags I can emulate with Flipper.

**Describe the solution you'd like**
Add a Write option for the different NFC protocols.

**Describe alternatives you've considered**
Emulating from Flipper is a partial workaround

**Additional context**
I'm personally interested in Mifare Classic, but unless it's drastically different effort between the protocols (which it may well be) it seems to make sense to track the overall feature under one issue.

GMMan, 2022-07-02:
One consideration for implementation is how you are going to handle writing to a tag sized differently than the original. Due to lock bits and other control options, this is not trivial and requires positively identifying the tag type and whether the data is fully writeable before attempting to write the tag.

skotopes, 2022-07-10:
Hi. We are currently working on it.

simkard69, 2022-07-28:
Thanks for working on that Feature Request, greatly appreciated !
Cheers from France !

Light-Wing, 2022-08-01:
Just wanted to do this and found that I couldn't...
But I can from my phone (quite a few nfc apps available for android)... 
Is there any ETA for this to happen soon or is it planned for a further down the road update?

skotopes, 2022-11-04:
@colinborins MFC cards write are already in release and next release will add mifare magic card

colinborins, 2022-11-04:
@skotopes, it seems that I can only write to the same tag and not a new one though, is that going to change?

skotopes, 2022-11-05:
Yes. Next release will include mfc magic card support.

FFX721, 2022-11-18:
@skotopes will it also be possible to write to an new card? <3

superyarik, 2022-11-21:
@skotopes is where is another NFC types write ability planned? NTAG213 f.e.?

skotopes, 2022-11-22:
Yes.

skotopes, 2022-12-01:
Yes, other generations of magic cards coming in future releases. I'll close this issue, if you need something specific please create new issue.

closed 2022-12-01 by skotopes

patrikcze, 2023-02-06:
Writing NTAG215 Planned? 👍