flipperdevices/flipperzero-firmware #1355: Extended menu shortcut settings/scripting

FalsePhilosopher, 2022-07-01:
**Is your feature request related to a problem? Please describe.**
It would be nice to be able to set custom shortcuts for every button including long press and on the fly create shortcut scripts IE Run sghz open garage door.sub>Run sghz unlock car.sub or Run badusb hotdogwallpaper.txt> Run badusb rickroll.txt 
**Describe the solution you'd like**
Options for shortcuts to menu's, submenus, and applications, an app in the shortcut menu that gives you options to let you build a script by selecting functions and files and what not. 

**Describe alternatives you've considered**
NA
**Additional context**
NA

skotopes, 2022-07-10:
Hi. We'll return back to this issue after v1.0.0.

FalsePhilosopher, 2022-07-14:
@skotopes While we are talking about menu extensions, can we also get a submenu in the power menu to change the time on how long the back button power menu shortcut takes? 1 being a long press, 3 and 5 like it is now? 5 is way too long for my taste, it should be set to 5 sec stock but the option to change that would really be nice.

skotopes, 2022-12-01:
After second thought looks like it will be impossible to implement it. We'll drop this issue from backlog.

closed 2022-12-01 by skotopes