flipperdevices/flipperzero-firmware #1394: BadUSB ALTSTRING outputs numbers

KairuByte, 2022-07-13:
**Describe the bug**
`ALTSTRING This line was print using Alt+Numpad input method. It works even if non-US keyboard layout is selected` outputs the following on latest firmware: 
```
784104105115321081051101013211997115321121141051101163211711510511010332651081164378117109112971003210511011211711632109101116104111100463273116321191111141071153210111810111032105102321101111104585833210710112198111971141003210897121111117116321051153211510110810199116101100
```

**To Reproduce**
Simply run the above in a BadUSB script on WIndows 11.

**Expected behavior**
This string: `This line was print using Alt+Numpad input method. It works even if non-US keyboard layout is selected`

**Target**
Windows 11

djsime1, 2022-07-15:
Looks like those are the raw alt codes for the text. My best guess is that Flipper failed to activate num-lock.

GeorgeBroughton, 2022-07-16:
it fails to hold ALT

Numlock is successful, ALT is not
it's typing the numbers fine

Can confirm this issue on a UK keyboard layout if this is relevant.
It could also be to do with timings of keypresses.

Knowing how Windows behaves when making USB keyboard/mouse emulators in the past, windows can do some buggy things if you type fast. Like if you use one to type your password at a windows login screen, winlogon.exe will actually crash.

So if ALT is being pressed too quickly before the numbers are being pressed, it might cause some issues.

Haven't looked at the code for it yet but this is my assumption as to what's wrong.

PR #1406 (2022-07-17): Update bad_usb_script.c to fix incorrect ALT key const
# What's new

- fix incorrect constant for ALT key causing incorrect behaviour of ALTSTRING and ALTCHAR functions in bad_usb scripts. Useful for non-US keyboard layouts

- Fixes #1394 

# Verification 

- Test scripts using ALTSTRING to input non letter keys on a bad_usb script

# Checklist (For Reviewer)

- [x] PR has description of feature/bug or link to Confluence/Jira task
- [x] Description contains actions to verify feature/bugfix
- [x] I've built this code, uploaded it to the device and verified feature/bugfix
closed 2022-07-17 by skotopes