flipperdevices/flipperzero-firmware #1403: CC1101 RX FIFO can sometimes duplicate bytes

LeeHolmes, 2022-07-15:
I was just reading the CC1101 Errata, and it looks like the Flipper Zero handling of the CC1101 RX FIFO doesn't currently incorporate the recommendation that TI makes to work around a chipset bug:

[image]

Link: https://www.ti.com/lit/er/swrz020e/swrz020e.pdf?ts=1657906815283&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FCC1101

Here is the current implementation of interaction with the RX FIFO in the CC1101, which doesn't include any polling code:

https://github.com/flipperdevices/flipperzero-firmware/blob/4d6b170769a33711e62deaf1c2a6b3bdda37e665/lib/drivers/cc1101.c#L154

hedger, 2022-07-20:
@Skorpionm

skotopes, 2024-01-03:
@Skorpionm would you kindly?

Skorpionm, 2024-01-12:
Our code is not affected by this errata. We use furi_hal_subghz_rx_pipe_not_empty to determine data availability, which checks RX_FIFO status flags instead of remaining byte count.

closed 2024-01-12 by skotopes

selenologist, 2024-04-22:
Hi. Sorry to reopen this, but it was closed with the wrong justification.

The issue is not the validity of the remaining byte count. Instead, the issue is the internal pointer of the device's FIFO. If the race condition occurs, the byte is duplicated from the perspective of the host because the FIFO has not advanced. This is not to be confused with the SPI Read Synchronization Issue.

Therefore, it would seem that the code IS affected by this errata. TI's workaround seems to involve knowing the number of expected bytes in the packet and making sure you don't fully empty the FIFO until the packet ends, which, ehh, doesn't really work if you don't know how many bytes are coming. 

I came here searching for examples of how real world code responded to the errata before using CC1101 in my own project and just noticed the inconsistency. I'm only drawing attention to it because, as the issue is a race condition, this may be the root cause for some rare issue. I don't want someone debugging this later to be like "oh, someone said we're not affected, it's fine" - maybe this could use another look. That's all. To be clear, I have not analyzed this fully myself. I got lost trying to work out what exactly happens here so maybe there's some other reason it's not affected.

Have a nice day.

skotopes, 2024-04-23:
> To be clear, I have not analyzed this fully myself. I got lost trying to work out what exactly happens here so maybe there's some other reason it's not affected.

[image]

selenologist, 2024-04-23:
What are you trying to say?

selenologist, 2024-04-23:
> Our code is not affected by this errata. We use furi_hal_subghz_rx_pipe_not_empty() to determine data availability, which checks RX_FIFO status flags instead of remaining byte count.

This does not solve the race condition when the FIFO is emptied while a byte is received.

I don't know if it's fixed somewhere else because I don't understand the entirety of the code here. I'm just pointing out the logic error. But I did take a quick look and I can't find any actual check to make sure the FIFO is not emptied when a byte may be being clocked into the FIFO, which is the crux of the bug.

I came here from a web search. I don't have a Flipper Zero. My intention is to prevent anyone else from getting confused by this issue being dismissed incorrectly. If the response is just going to be confusing images then, I think I'm done here. I've done my part.