flipperdevices/flipperzero-firmware #1411: Can't read SD card after using 3V3 port

thisisALPINE, 2022-07-17:
**Bug**
Whenever the 3V3 port is used to supply current to some hardware, upon coming back the Flipper has issues using the SD card and consequently cannot read anything out of it until rebooted.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on the Flipper
2. Attach hardware to 3V3 port
3. Try using the Infrared Remote or whatever thing that uses the SD card
4. Receive error message "Cannot create app folder" or "No databases found on SD card"

**Expected behavior**
I expect the Flipper to keep using the SD card while supplying current to external hardware

**Target**
In this case I used an ESP32 connected to the pins GND and 3V3 of the Flipper.

[image]
[image]
[image]

skotopes, 2022-07-17:
That is expected behavior.

tomholford, 2022-12-16:
Having this issue after the latest update, but it seems to occur randomly (ie., not after using the 3v3 port)

skotopes, 2022-12-16:
@tomholford Try to format sd card

skotopes, 2023-06-22:
Was fixed some time ago.

closed 2023-06-22 by skotopes