flipperdevices/flipperzero-firmware #1416: Misspelled Item in `3164184bb` - application/subghz/views/subghz_read_raw.c

wyattearp, 2022-07-19:
# What are you want to add or change
It looks like 3164184bb introduced a typo of changing `status` --> `satus`

# What questions do you have
It should probably be `status` :)

```c
// application/subghz/views/subghz_read_raw.c
typedef struct {
    string_t frequency_str;
    string_t preset_str;
    string_t sample_write;
    string_t file_name;
    uint8_t* rssi_history;
    bool rssi_history_end;
    uint8_t ind_write;
    uint8_t ind_sin;
    SubGhzReadRAWStatus satus;   <-- here
} SubGhzReadRAWModel;
```

Running this should fix it (from the `subghz` folder:
`find . -type f -exec sed -i 's/satus/status/g' {} \;`

Skorpionm, 2022-07-26:
😳

wyattearp, 2022-07-26:
that's weird - why didn't it close with the message? -- well, it should have. let me fix that and thanks @Skorpionm :)

closed 2022-07-26 by wyattearp