flipperdevices/flipperzero-firmware #1434: RC5 emuation not working correctly on Grundig TV

Leseratte10, 2022-07-23:
**Describe the bug**
Tried to emulate my TVs IR remote with the Flipper, didn't work

**To Reproduce**
Steps to reproduce the behavior:
1. Tried to clone an RC5 IR button
2. Emulate said button
3. Notice that the TV didn't do anything

**Expected behavior**
TV does something, i.e., the key is emulated correctly.

**Logs**
Excerpt from the IR file generated by Flipper: 
```
Filetype: IR signals file
Version: 1
# 
name: Power
type: parsed
protocol: RC5
address: 00 00 00 00
command: 0c 00 00 00
# 
name: Num_1
type: parsed
protocol: RC5
address: 00 00 00 00
command: 01 00 00 00
# 
name: Num_2
type: parsed
protocol: RC5
address: 00 00 00 00
command: 02 00 00 00
#
```

[image]

**Target**
Target TV is a Grundig Davio 20 apparently using the RC5 protocol.

**Additional context**
Is there any way to force the Flipper to record IR RAW signals, even if it thinks it detected a particular protocol? Maybe that would help, but so far i haven't figured out how to do that or if that is possible.

skotopes, 2022-09-06:
Do you have any other tools to record raw? 

@gsurkov I guess we'll need raw recording for cases like this one.

Leseratte10, 2022-09-06:
I don't, no. I have a bunch of Arduinos laying around, but I don't think I have one of these IR receiver modules. If necessary I can buy one as they seem to be fairly cheap. I assume the Flipper currently doesn't have a way to force a raw recording?

skotopes, 2022-09-06:
Nope it doesn't. But looks like it is something that we going to add next

gsurkov, 2022-10-14:
Looks like the raw capture option needs to be implemented. Will work on it soon.

gsurkov, 2022-11-08:
@Leseratte10 it is now possible to record raw signals via the CLI. 
In case the problem is still relevant, consider doing the following:
1. Run `ir rx raw`, send the signals and save the output.
2. Create a remote file and add these signals to it (use another remote as reference).
3. Send these signals to the TV via Flipper and check if they do what they are supposed to.
4. Post the raw signals and the TV experiment results here.

Leseratte10, 2022-12-11:
Sorry for the late response - I just tested "ir rx raw". I wasn't sure what to put for "frequency" and "duty_cycle" as that wasn't printed with "ir rx raw" so I just copied the values from another remove (38000 and 0.330000). 

This remote seems to be using a unique solution to prevent the TV from recognizing a single button press twice ...
Here's five recordings of the POWER button:
```
RAW, 47 samples:
852 848 1669 845 8 849 822 853 829 847 824 851 830 845 826 850 832 1681 826 850 1667 846 825 85452 
849 851 1666 847 824 851 830 845 826 849 832 843 828 847 824 851 830 1683 824 852 1665 848 823 
RAW, 47 samples:
854 847 824 851 1666 847 824 851 830 845 826 850 832 844 827 848 823 1690 827 848 1669 844 827 85450 
851 848 823 852 1665 848 823 853 829 846 825 851 830 845 826 849 832 1681 826 849 1668 845 826 
RAW, 47 samples:
848 853 1664 849 822 853 829 847 824 851 830 845 826 849 832 843 828 1685 832 843 1664 849 832 85445 
855 846 1660 852 829 846 825 851 830 845 826 849 832 843 828 847 824 1689 828 847 1670 843 828 
RAW, 47 samples:
855 845 826 850 1667 846 825 850 832 844 827 848 823 853 828 847 824 1689 828 847 1670 843 828 85449 
852 848 823 853 1664 849 822 853 828 847 824 851 830 845 826 850 832 1681 825 850 1667 846 825 
RAW, 47 samples:
848 853 1664 849 832 843 828 847 824 851 830 845 826 850 831 844 827 1686 831 844 1663 851 830 85446 
852 849 1668 845 826 850 831 843 828 848 823 852 829 846 825 851 830 1682 824 851 1666 847 824
```

So it looks like each button press sends two signals with a short pause in between (the 854xx value), but also the third and fifth value are swapped with each button press. 

I tested this with the Flipper and first I thought (wrong assumption): "Hey, cool, the 1st command always turns ON the TV, the 2nd one always turns it OFF, what a cool feature". 

Then I noticed that even buttons where that toggle thing doesn't make sense have two different recordings (like the "1" button, and probably all other buttons too). 

It looks like the first five-ish values that are changing between button presses are a global counter to detect duplicate keypresses. Lets name the prefix `855 845 826 850 1667` as "A" and the other one (`848 853 1664 849 832`) as "B". 

So once the TV receives any button input for any button where the raw signal starts with "A", it will no longer respond to *any* commands starting with "A". The next command sent to the TV will *have* to use the "B" command prefix. Then after that the TV ignores the "B" prefix and the next command needs to start with "A" again for the TV to recognize it. 

That means that to support this weird TV, the Flipper would need to have a global bit of storage for each IR file, so the flipper knows whether the next button press should use prefix A or prefix B. 

I tried being sneaky and putting the RAW data for the power button with prefix A and for the power button with prefix B right after another, hoping that the TV would use one and just ignore the other, but that didn't work either. 

If I put them right after another without a pause, the TV ignores it. If I put a pause, the TV will register two POWER button presses and turn on, then immediately turn off again.

If I don't just push and release a button while recording with the Flipper but I hold down that button instead and keep it pressed, the remote will send the same command multiple times with the same Prefix value. 

Not sure if this is something that can be implemented into the Flipper somehow or if that's going to be too complicated ...

skotopes, 2024-01-03:
@Leseratte10 I'll close this issue. Whole thing sounds quite exotic and requires major changes in architecture.

closed 2024-01-03 by skotopes