flipperdevices/flipperzero-firmware #1436: BadUSB Example on Apple fails

jeroenh, 2022-07-23:
**Describe the bug**
The example badusb script for apple devices should use spotlight to open a terminal.
The enter key is not sent correctly, so all the input is entered into the spotlight search window instead of a terminal window

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on...
2. connect flipper to a usb-port on a Mac
3. Run the BadUSB example script

**Expected behavior**
Terminal opens, and some nice graphics and text are entered to see what can be done with badUSB scripts

But the return is not sent correctly after 'cmd-space terminal', causing all text to be entered into the spotlight search bar.

SebCorbin, 2022-07-26:
Relying on `cmd`+`space` is also messy because it may be configured to use `ctrl`+`space` in Keyboard shortcuts

nohehf, 2022-07-26:
Isn't there any way to start the terminal without relying on spotlight?

SebCorbin, 2022-07-28:
A solution to this would be to hit both `cmd`+`space` and `ctrl`+`space` as they do not interfere once spolight is open (from either shortcut)

jeroenh, 2022-08-09:
latest version of the firmware 0.63.3 is able to launch terminal properly and start `cat` and pushing text.

closed 2022-08-09 by jeroenh