flipperdevices/flipperzero-firmware #1441: Use custom NFC dictionaries based on their UID for known Mifare cards

der-pepe, 2022-07-24:
The nfc_worker_mifare_classic_dict_attack function takes quite some time to iterate along all the key on the mf_classic_dict.nfc file. For new cards this is just fine, since we don't actually know its keys, but for known well known cards, this is quite tiresome.

A quicker method could be, before loading the generic dictionary file, to check for a custom one (maybe something in the format of {{CARD_UID}}.nfc) so it would load only the specific keys for that card.

This would help keeping the original dictionary clean, and speed up greatly the process for known cards, and in the future, even some kind of configuration mapping between multiple UIDs and a shared dictionary file could be implemented.

Astrrra, 2022-07-24:
A similar optimization is already present in the draft NFC-redesign branch, so this will be solved once that branch gets finished and merged into dev

tkerby, 2022-07-27:
So I'm trying some of the new NFC branch on the RogueMaster firmware. My comments

1 - The cache shouldn't be per card. It would be great to be able to read my own facility card for instance then instantly read others. That either means an option should be there to view the cached keys and add them to the user dictionary or the cache files should be read quickly on each card to check for matches

2 - One of the big slowdown issues in reading cards is that if I add keys to the user dictionary, then for each sector, the flipper tries through the list from top to bottom. So if I have 1000 keys in the dictionary and mine is last for sectors 1 and 5, then for sector 1 the flipper will try 1000 keys and also for sector 5 it will try 1000 keys. We should cache keys while reading a card and make them the first keys tried against any additional sector

simkard69, 2022-07-28:
This has be done in RogueMaster firmware if I'm correct 👍

tkerby, 2022-07-28:
> This has be done in RogueMaster firmware if I'm correct 👍 

Yes - at the moment it does as they've pulled the development branch of FZ which includes this

Astrrra, 2022-12-08:
Is this still relevant? Most of the improvements described above have already been implemented.

closed 2023-03-13 by skotopes