flipperdevices/flipperzero-firmware #1442: Import frequency from frequency analyzer to read raw

Project-Code-Name, 2022-07-24:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

The workflow would be this: Open the frequency analyzer, get the frequency the device transmits on, you now have the option to either throw away that frequency or you can "import" it to read raw where it would then pre-set that frequency.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

A way to keep the found frequncy from the analyzer on-screen till the user presses a button.

**Additional context**
Add any other context or screenshots about the feature request here.

Skorpionm, 2022-07-26:
What if the remote control is multi-frequency?  and if the frequency stabilization at the remote control floats?  how to be then?

pmarrapese, 2022-07-27:
I opened a related feature request that could incorporate the desired functionality in an elegant way: #1466

closed 2023-03-13 by skotopes