flipperdevices/flipperzero-firmware #1457: HardFault trying to use saved Sub-Ghz

SharpEdgeMarshall, 2022-07-26:
### Describe the bug.

Every time I try to "Emulate" a Sub-Ghz remote that I saved before, Flipper Zero power off and I have to use the reset sequence to restart it (a message will appear saying that FlipperZero was restarted cause of a HardFault)

### Reproduction

1. Go to Sub-Ghz
2. Read remote (CAME TWEE) on 433.92 AM270
3. Save it or press send (if you press "Send" Flipper zero will shutdown)
4. Go to saved
5. Select saved and press "Emulate"

### Target

Remote model: ENTREMATIC ZEN4

### Logs

```Text
How do i find logs?
```

Skorpionm, 2022-07-26:
https://github.com/flipperdevices/flipperzero-firmware/pull/1446
if you really use the CAME TWEE remote control, there will be no more errors.  if not this protocol, then it will not be identified, the error was due to an incorrect definition of the protocol

skotopes, 2022-07-26:
fixed in 0.63.0-rc

closed 2022-07-26 by skotopes

SharpEdgeMarshall, 2022-07-26:
@skotopes is detected as CAME TWEE but is a ENTREMATIC ZEN4 how can I know if it is normal?

skotopes, 2022-07-26:
@SharpEdgeMarshall if emulation works then it's normal )

SharpEdgeMarshall, 2022-07-27:
@skotopes I tried the `0.63.1` and the issue is still there :/

Skorpionm, 2022-07-27:
then please write me a raw file with this remote control and send