flipperdevices/flipperzero-firmware #1463: RFID Reading/Writing fails on specific tag

Pingvin0, 2022-07-26:
### Describe the bug.

I ordered some (what I assume to be) T5577 tags.
When they arrived and I tested them, the flipper zero could not read or write to them.
I started adding debug logging statements to the 0.62.1 firmware, and after a while I noticed reading suddenly worked.
I narrowed it down to one change I've made:
In `RfidReader::read` after the `if(something_read)` statement, I added an else condition which writes a debug log message.
This is my code modification:
```cpp
    // validation
    if(something_read) {
        switch_timer_reset();

        if(last_read_type == *_type && memcmp(last_read_data, data, data_size) == 0) {
            last_read_count = last_read_count + 1;

            if(last_read_count > 2) {
                result = true;
            }
        } else {
            last_read_type = *_type;
            memcpy(last_read_data, data, data_size);
            last_read_count = 0;
        }
    // from here
    } else {
        FURI_LOG_D("LFRFID", "Nothing read");
    }
    // to here
```

When logging is set to debug, this allows the blank tags to be read.
I suspect it is a timing issue.

### Reproduction

Switch on the flipper, select 125 KHz RFID->Read and hover the tag.

I have also made a video on what happens: [https://streamable.com/l1bzms](https://streamable.com/l1bzms)

### Target

Tested on 0.62.1 and c7772060657afc8ee5a160ea3ed62246d989c136

hedger, 2022-07-27:
@DrZlo13

skotopes, 2022-09-06:
Please check latest release

skotopes, 2022-09-06:
And reopen if issue persist

Pingvin0, 2022-09-07:
Hello. Tested on 0.66.1 and d1c79a83dea7a7be7f36eb14c186926dda498a6d.
Issue still persists, reading/writing does not work on these specific tags.

Pingvin0, 2022-09-07:
@skotopes cannot reopen since you closed it

skotopes, 2022-09-07:
@DrZlo13 @nminaylov

doomwastaken, 2023-06-23:
works on all available tags, no other reports, closing an issue

closed 2023-06-23 by doomwastaken