flipperdevices/flipperzero-firmware #1465: Sub-GHz scan is missing PMR frequencies (EU version)

rsplaul, 2022-07-27:
### Describe the bug.

The Sub-GHz scan of the Flipper Zero (EU version) cannot be tuned to [PMR446 frequencies](https://en.wikipedia.org/wiki/PMR446) as used by restaurant pagers like this one:
[image]

### Reproduction

1. Select `Sub-GHz` in menu
2. Select `Read`
3. Click left for `Config`
4. Try to select Frequency

Skorpionm, 2022-07-27:
you can add this frequency to subghz/asset/setting_user

rsplaul, 2022-07-27:
> you can add this frequency to subghz/asset/setting_user

I was under the impression, that the Flipper firmware only allows for scanning frequencies from the [Table of frequencies by region](https://docs.flipperzero.one/sub-ghz/frequencies). PMR446 isn’t on that list at all.

Also, I do not understand the syntax of the `setting_user` file. Is there any documentation about this file?

theobarrague, 2022-07-27:
This is mine for example :
```
Filetype: Flipper SubGhz Setting File
Version: 1

# Add Standard frequencies for your region
add_standard_frequencies: false

# Default Frequency: used as default for "Read" and "Read Raw"
default_frequency: 433920000

# Frequencies used for "Read", "Read Raw" and "Frequency Analyzer"
frequency: 315000000
frequency: 433920000
frequency: 446156250
frequency: 868350000

# Frequencies used for hopping mode (keep this list small or flipper will miss signal)
hopper_frequency: 315000000
hopper_frequency: 433920000
hopper_frequency: 868350000
```
File is self-documented

rsplaul, 2022-07-27:
Maybe I am doing something wrong, but changing the settings file does not have any effect at all. I also tested with the exact file posted above by @theobarrague.

theobarrague, 2022-07-27:
Did you change the file under SD Card / subghz / assets / setting_user ?

I just updated my firmware with the latest version on dev branch and I use this :
```
Filetype: Flipper SubGhz Setting File
Version: 1

# Add Standard frequencies for your region
Add_standard_frequencies: false

# Default Frequency: used as default for "Read" and "Read Raw"
Default_frequency: 433920000

# Frequencies used for "Read", "Read Raw" and "Frequency Analyzer"
Frequency: 315000000
Frequency: 433920000
Frequency: 446156250
Frequency: 868350000

# Frequencies used for hopping mode (keep this list small or flipper will miss signal)
Hopper_frequency: 315000000
Hopper_frequency: 433920000
Hopper_frequency: 868350000
```
( I don't know if the case matters )

vr, 2022-07-28:
i have the same issue @rsplaul , i can't change frequencies anymore in 0.63.1-RC or dev version, it works fine with 0.62.1, i'm wondering if this could be related to preset refactoring done by @Skorpionm ? i'm digging the code

bbock, 2022-07-28:
@rsplaul @vr capitalization of the first character in `setting_user` changed recently, e.g. `Frequency:` vs. `frequency:`. Check the example [here](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/assets/resources/subghz/assets/setting_user).

vr, 2022-07-28:
yes just saw this ! Thank you @bbock

Skorpionm, 2022-08-01:
we bring everything to the same standard, and now in the settings, as in the saved key files, all parameter names begin with a capital letter. It's better to do it now before it really hurts. if your problem is solved, close the issue

rsplaul, 2022-08-27:
Sorry for leaving this open for so long. I can indeed confirm that is was about the capitalization in the settings file.

closed 2022-08-27 by rsplaul