flipperdevices/flipperzero-firmware #1466: History of detected frequencies in Frequency Analyzer

pmarrapese, 2022-07-27:
### Describe the enhancement you're suggesting.

**Is your feature request related to a problem? Please describe.** 
The Frequency Analyzer is great for reconnaissance and finding frequencies used within the area. However, the analyzer only shows a detected frequency for a second before resetting to 000.000 MHz. Blink and you'll miss it.

**Describe the solution you'd like.**
It would be useful if the analyzer kept a list of detected frequencies (along with their RSSI, and maybe a timestamp). It could be in a submenu so the current interface is preserved, but the user could press a button to switch to a list view of everything that's been detected. From there, maybe there could even be a "Send to reader" action to switch into Read mode for that frequency.

**Describe alternatives you've considered.**
I know Read mode has a "Hopper" function, but that seems to be more intended for capturing signals from specific freqs that are using recognized protocols.

**Additional context:**
My use case here is checking out which freqs are used by garages in busy downtown areas. The RSSI history would be great for helping to figure out the signal origin, too!

Skorpionm, 2022-08-01:
we bring everything to the same standard, and now in the settings, as in the saved key files, all parameter names begin with a capital letter. It's better to do it now before it really hurts. if your problem is solved, close the issue

skotopes, 2022-10-16:
Available in dev, will be release in 0.69

closed 2022-10-16 by skotopes