flipperdevices/flipperzero-firmware #1468: NFC Emulation of previously saved Mifare classic 1k card does not work.

Pingvin0, 2022-07-27:
### Describe the bug.

I have multiple saved cards from 0.62.1, and in the latest release candidate, when I emulate the card, my phone cannot detect it at all. When I emulate a saved NTAG/Ultralight there is no issue.

### Reproduction

On 0.62, save a Mifare classic 1k card
Update to RC
NFC->Saved->saved card->Emulate
Phone cannot detect emulated card

### Target

release-candidate

### Logs

```Text
This is for a wifi tag I had: https://pastebin.com/R8s1gY2L

And this is for a contact (VCARD) mifare classic 1k card: https://pastebin.com/vA5HdcwK
```

gornekich, 2022-07-27:
Thanks. Working on fix

Pingvin0, 2022-07-27:
Just tested this on c7772060657afc8ee5a160ea3ed62246d989c136
Reading a brand new MFC card with content and emulating it fails as well. I don't think it's about previous save data being wrong.

gornekich, 2022-07-28:
The reason can be enabled log level debug in Settings->System. Please confirm that you set log level to Info / Warning / Error / Default

simkard69, 2022-07-28:
I have the same problem on my side, yesterday tried to brute-force attack a Mifare-classif ... which succeeded after 15 minutes.
Anyway, once I tried to emulate the tag in front of the reader : not working (it was a VIGIK like this one : [VIGIK READER](https://cdn.adiglobal.fr/userfiles/FullRepository/Large/T25VK3_lg.jpg))

I have some blanks NFC Mifare-Classic cards (white ones with UID/Sector 0 writable), but faced the fact that there are no ways to write NFC tags at all for now in the Flipper Zero firmware.

Linzdigr, 2022-07-28:
Exactly in the same situation as @simkard69 : NFC Mifare Classic 1k card's keys and content are well extracted by the Flipper (👍) but the emulation seems to be the issue. The VIGIK denies the emulated version.

From what I've understood it's a timing issue with the Flipper's NFC chip onboard not natively supporting the Mifare Classic 1k (and the current soft implementation is leading to timeout with the VIGIK communication), right ?

Anyway, hope software workaround will be enough to go through this issue !

Xenthys, 2022-07-29:
Following commit c40e8811, I have been able to emulate a Mifare Classic 1k card successfully!
I've been able to read it with NFC Tools on my phone, so now if it fails I'm going to blame the reader.

Thanks for the fix @gornekich, you may be waiting for other confirmations but in my case we're good!

Pingvin0, 2022-07-29:
@Xenthys I still cannot emulate freshly scanned or saved MFC 1k cards on c40e8811d68e9f4b8f603ae5d5826b814521014d.

Linzdigr, 2022-07-29:
Flipper firmeware changed from stable to dev https://github.com/flipperdevices/flipperzero-firmware/commit/c40e8811d68e9f4b8f603ae5d5826b814521014d

Results:
- Previous saved MFC 1k file emulated -> doesn't recognized by the VIGIK reader
- Fresh scanned and keys retrieved then emulate -> still KO

Thanks for the update though ;)

BestPig, 2022-07-29:
Still have some issue at some point for me.

Unable to read the whole contents with MIFARE Classic Tool (Android)
And also with an ACR122U with the libnfc

It seems the tag disappears during the read.

Xenthys, 2022-07-29:
All my cards work so this is odd, maybe try with one of mine / compare the content?

This one is safe, it's an old metrovalencia ticket from months ago:
https://gist.github.com/Xenthys/4a7cc356b1a1caf052c9986495fb9c46

goutchye, 2022-08-11:
Hi guys!

I have also tried to emulate mifare classic cards on a vigik reader, with no success.

To check if the flipper zero emulation was working I have also tried to "read" it using a mfrc522 rfid reader, it was a failure except for the first sector which was good. I discovered that I could read each sector with success but never read many of them in a row.

SO, I have tried to read a delay between each reading attempt and it was working when I added a delay of 700ms.

Do you think this it is a hardware limitation of the flipper zero or "just" a software issue?

I hope that maybe this information would help.
To finish, thx all of you for the amazing job on the project :-)

theblackhole, 2022-09-25:
Any news on this issue ?
With 0.67.2, I still can't get Mifare Classic 1k tags to work with Vigik readers (and some parts are missing when I read my emulated badge with Mifare Classic Tool)

Coroxx, 2022-09-25:
@theblackhole yes. When you put your badge on the VIGIK READER, the reader check if your badge is correct, it increments a value, and after he check if the value was incremented.

So when you put your flipper on the vigik reader, it fails on the second step because you cannot write on the flipper during emulation. 

This double security is only made by VIGIK.. The most of the other brands check only the badge. 

I hope you understand what i'm trying to explain.

simkard69, 2022-09-25:
I have some dumps of universal VIGIK badges from LaPoste.
They have to be re-writed every 3 to 4 days in order to continue working.

I'm pretty sure the reader cannot use that much encryption/cyphers in order to do what it needs.
Should I try to dump a badge, use the VIGIK reader, then dump it again (... and so on), in order to find where is the increment number located on the memory ?

goutchye, 2022-09-25:
@Coroxx you are right but the value is incremented only if the badge is a "service provider" badge (postmail, electricity, gaz, etc.). In this case, the provider must encode a badge which will only be valid during 3.5 days as @simkard69 has mentioned, and the counter is probably useful to limit the use of the badge.

In the other case, if it is a "home" badge for people who live in the building, only the UID, or the first sector is used.

Each VIGIK central can work with both cases.
FIY, the emulation of my own "home" badge fails on the vigik central of my building (test with the firmware released the 9th of august).

The "best" way to find an issue would be to sniff (thanks to a proxmark) the timing of the exchange between a vigik central and a valid badge, and try to reprocuce them on a flipper zero (who seems to be a little bit "slow" for the moment).

Coroxx, 2022-09-25:
So is it a software or a hardware issue ?

theblackhole, 2022-09-25:
Interesting... In my case, it is also a resident badge, not a service card. 

In Mifare Classic Tool, the first sector is always read correctly. For other sectors, missing data occurs randomly (my last test result was a missing key on sector 14, and on the previous test result it was sector 10)
So if only the first sector is used it should work in theory, unless Vigik readers are more finicky than smartphones ?

Also FYI, I also have an old UID-only dump of my badge (from a previous firmware) wich doesn't work either. And I compared my badges dumps from Mifare Classic Tool with the flippers and they are identical. So it seems like only the emulation mode might be affected by timing issues.

Is there a way to get logs of what is happening during emulation (or maybe with "Detect Reader" mode ?) ? Because while buying a Proxmark is tempting and would satisfy my curiosity, I think it is quite pricey for a beginner/hobbyist like me ^^

goutchye, 2022-09-26:
@Coroxx I don’t know.
I hope it is « only » a software limitation.

@theblackhole yes in theory it « should » works. It also depends of the data writing on you badge. Is there data only in the first sector ? 

For the logs questions, I don’t know.
I also think than adding too much logs could slow down a little bit the flipper zero no?
Yeah the proxmark is expensive but it is probably the best way to debug this problem.

I Hope I could try it in few weeks or months.

theblackhole, 2022-09-28:
> @theblackhole yes in theory it « should » works. It also depends of the data writing on you badge. Is there data only in the first sector ?

No, not only the first sector. In a valid dump (of my badge with Mifare Classic Tool directly) there's data on the first 4 sectors. For sectors 5 through 15, it's filled with zeros except the last block of each sector where keys are stored. (In my case, it's the same key everywhere).
If I wasn't clear, I can post a "censored" version of my dump to clarify my point if you want.

While exploring the new features of 0.62.2, (keys extraction with mfkey32v2 in particular) I found out that there's a way to display flipper's logs via usb : https://github.com/equipter/mfkey32v2#using-log
I'll see if I can make something out of it... but before that, I need to find a discrete setup because a pc is a little more noticable than the flipper hahaha
Also, as you said, it might make things worse by slowing down the execution but we'll see.

AkechiShiro, 2022-10-18:
@theblackhole any news on this, have you found anything relevant using the flipper's logs via USB ? 

On this issue, I think it's important to mention a few things : 
- Flipper Zero must not be in "Settings" -> "System" -> "Debug ON", if it is activated, reading any Emulated tags will always fails (tested on the latest stable release 0.69), reader was an Android Phone with NFC, make sure "Debug" (mode) is set to "OFF". (A PR into the documentation of Flipper Zero could help, I could make one, mentioning this).

About, a Mifare Classic 1k card, that I'm trying to emulate and get to work with a VIGIK reader.

To sum up, what's working on my end:
- [X] Reading the physical tag with my Flipper Zero (attack by dictionary works all right).
- [X] dumping with my phone the emulated tag from the Flipper Zero (previously acquired using read).

However, I've noticed a slight difference on sector 15 of my dump, emulated tag dump vs the real physical tag dump, I believe a specific block of the sector 15 is getting incremented by one.

This seems to be an anti-copy/duplication protection, that's updated by the VIGIK reader, each time the physical tag is used, **a desync of the physical tag could happen, if the Flipper Zero emulation worked for me** as mentioned in the Flipper documentation, the emulation needs to take into account, writes request, so that the tag could be modified *in place*.

That's unexpected given the nature of my tag, it is not a "service provider" one, it works without being updated for weeks and months on.

Hugal31, 2022-10-19:
@AkechiShiro thanks for the Debug tip. It still cannot read the emulation with Mifare Classic Tool on my Fairephone 4 (it rarely detects it, and when it does it drops the connexion right away).

For the anti-copy feature of your tag, you can read about in #1345. The best test against it is what you did, scanning the real tag before and after using it and compare the two dumps.

AkechiShiro, 2022-10-19:
@Hugal31 you tried on the latest 0.69 firmware version and it stills failed to read it with the Mifare Tool app ? That's what I used and it works on a Xiaomi Poco F3

AkechiShiro, 2022-10-20:
Also @Hugal31 could you test [NXP TagInfo](https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=en&gl=US) in order to read your emulated tag on the Flipper Zero ?

Here is a [scan log in XML of an emulated tag](https://gist.github.com/AkechiShiro/cad49659342c5d8c2a6ab2af252719c4), I did, I redacted some information, but they were properly detected (UID, ATQA, SAK,...).

This does not try to dump the tag, but just tries to read a few information by scanning quickly (it takes on my phone about 10 seconds at most I believe, sometimes the communication is cut off if I move the Flipper Zero too much once the scan has begun, make sure you hold the phone and the Flipper Zero well together).

AkechiShiro, 2023-02-12:
I'd like to note an interesting observation that there are numerous VIGIK readers versions (so far, I'm aware of V1/V2/V3), could everyone that mentioned that the emulation didn't work, look at the version at the bottom of the reader and precise for which version it did fail ? 

This might help us, find maybe an old version of reader where emulation works still works with.

So far, looking around, I've seen V2 and V3 readers but not V1 readers, here is a picture of what it looks like : 
[image] 

@Coroxx V2 readers, can sometimes increment a value inside a sector of the badge as an anti-counterfeit protection (but they don't do ALL do that since I've seen a V2 VIGIK reader that does not increment the value and for which the FP Zero emulation still fails for, so this can't be due to some "write/increment" request failing on the emulation side).

I suspect that V2/V3 probably have some "features" or differences that V1 doesn't, I'm looking to know if a VIGIK V1 reader works with the Flipper Zero's emulation or not (of the latest version of the firmware 0.76.0).

rvalitov, 2023-02-13:
@AkechiShiro Sorry, I have never seen a version like V1 or similar. Just no extra info except the VIGIK name. Some examples are below, perhaps that's useful for you
[image]
[image]
[image]

dybman, 2023-04-13:
Hello,
Will emulation ever work for "Vigik" or is it simply impossible to achieve?

AkechiShiro, 2023-06-12:
@skotopes could we please have any official information on this issue, it's been awhile, I know that #2529 is a full refactor of the NFC stack of the FP Zero's firmware but I don't think there is any fix regarding the VIGIK issue.

If there is any way/investigation that you or @gornekich would recommend, any user tries using either a Proxmark RDV3 or using an SDR (Software Defined Radio) ?

So we can really hammer on whether this issue can be solved or not.

skotopes, 2023-06-12:
@AkechiShiro we currently don't have Vigik tags/readers. We were planning to revisit this issue after refactoring complete.

Astrrra, 2023-06-12:
Also, a full emulation trace from a Proxmark3 would definitely help. If someone can do it - we need four things:

1. A sniff of the communication between a reader and a genuine card.
2. A sniff of the communication between a Flipper on a working firmware version (0.62.1?) and a reader.
3. A sniff of the communication between a Flipper on a current firmware version (latest release).
4. A full card dump (can be acquired with `hf mf autopwn` on the proxmark)

Flipper's debug toggle in System settings should be disabled, log level should be either None or Default.
Assuming your Proxmark3 is on the latest Iceman firmware, a trace can be captured with `hf 14a sniff`, and saved with `trace save -f <filename>` (don't forget to stop the recording by pressing the button on your proxmark, otherwise the saved file will be empty).

For convenience, please name your trace files along the lines of `card`, `working_fw`, `latest_fw`, and `card_dump` (in bin, json, and eml formats), respectively. The file extensions for the traces will be added automatically for your proxmark client.

You can attach them to this issue, or, if the card contains private data, send them to me in an email: [astra@flipperzero.one](mailto:astra@flipperzero.one). Please avoid using filesharing platforms with time-limited links, as we might not be quick enough to get to your files before they expire :P

sfjuocekr, 2023-06-19:
[package.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11783241/package.zip)

I'm just trying to read the emulated card with an ACR122U, just ordered a ProxMark as well it will probably take a while before it is here!

sfjuocekr, 2023-06-26:
```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |       1056 | Rdr |26                                                                       |     | 
       2244 |       4612 | Tag |04  00                                                                   |     | 
      13184 |      15648 | Rdr |93  20                                                                   |     | 
      16820 |      22708 | Tag |d4  4d  27  0b  b5                                                       |     | 
      44016 |      54544 | Rdr |93  70  d4  4d  27  0b  b5  33  1f                                       |     | 
      55732 |      59252 | Tag |08  b6  dd                                                               |     | 
     188000 |     192704 | Rdr |e0  50  bc  a5                                                           |     | 
     195636 |     196276 | Tag |04                                                                       |     | 
     331232 |     332288 | Rdr |26                                                                       |     | 
     407136 |     408192 | Rdr |26                                                                       |     | 
     483040 |     484096 | Rdr |26                                                                       |     | 
```

The results of the trace, when I'm done raiding in WoW I will do some more traces and dump it on the Proxmark!

sfjuocekr, 2023-06-26:
[tracedump.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11872085/tracedump.zip)

Nevermind, got em!

PsychoBitchy, 2023-07-07:
> Also, a full emulation trace from a Proxmark3 would definitely help. If someone can do it - we need four things:
>
>     1. A sniff of the communication between a reader and a genuine card.
>
>     2. A [...]

I did What you asked, the 3 traces are IDENTICAL (no diff at all) 
Sent you a mail with everything.

AkechiShiro, 2023-07-18:
@PsychoBitchy could you post a photo of the reader please ?

I ordered a PM3 but my order was lost during delivery, I will try and see if I can find someone who can lend me one, I have access to v2 readers.

There is also something weird with the detection field, it seems to be a bit random to work with the Flipper Zero on a reader v2.

AkechiShiro, 2023-07-20:
PR #2825 in latest release 0.87 seems to fix the issue with all VIGIK readers I've tested so far (ReaderV2/V1 mostly)

Hugal31, 2023-07-21:
This PR make the emulation works better with Mifare Classic Tool on my Fairphone 4. There is no more error message, however I often have "dead sectors" or some bytes that are not read. I did a successfully complete read after multiple tries though.

However I could not make it work with a VIGIK reader (without number, presumably V1) so far.

goutchye, 2023-07-21:
Hi guys
Today I have updated my flipper with the last firmware, dumped a legitimate postal badge and tried to emulate it but it has failed. The reader was a comelit model (don't know the version), and of course the legitimate badge was working well.

theblackhole, 2023-07-23:
> PR #2825 in latest release 0.87 seems to fix the issue with all VIGIK readers I've tested so far (ReaderV2/V1 mostly)

Finally ! 🎉 My cloned resident badge now works on a V2 reader (Urmet). I'll try with another reader (V1 maybe ?) when I get the opportunity.

> Hi guys
> Today I have updated my flipper with the last firmware, dumped a legitimate postal badge and tried to emulate it but it has failed. The reader was a comelit model (don't know the version), and of course the legitimate badge was working well.

If you dump the badge twice with Mifare Classic Tool: one dump before using the badge and one dump after using the badge, do you get the exact same dump ?
If I remember correctly, I heard/read that postal badges, since they can open every Vigik doors, have additional security features like authorized access during a defined time slot, defined maximum number of use, anti-clone features (if your 2 dumps are different, this could be your case)

goutchye, 2023-07-24:
Hi,
Yes this is still the problem. The legitimate postal badge has an counter in its memory which is incremented by the vigik reader each time it is reading. So the vigik reader, before to opening the door, is probably trying to write this counter value but it fails and the door doesn’t open 😕

theblackhole, 2023-07-25:
> Hi, Yes this is still the problem. The legitimate postal badge has an counter in its memory which is incremented by the vigik reader each time it is reading. So the vigik reader, before to opening the door, is probably trying to write this counter value but it fails and the door doesn’t open 😕
Ok that's what I thought. In this case, I'm glad this security measure exists and I'm glad this still doesn't work for you (sorry ^^) otherwise it would be a critical national security issue :) ...unless LaPoste badges are restricted to a defined geographical area (like one or two cities)? (but that would surprise me)

simkard69, 2023-07-25:
> Ok that's what I thought. In this case, I'm glad this security measure exists and I'm glad this still doesn't work for you (sorry ^^) otherwise it would be a critical national security issue :) [...]

Is there any way to allow F0 to simulate a badge with memory being able to be modified ?

Hugal31, 2023-07-25:
> Is there any way to allow F0 to simulate a badge with memory being able to be modified ?

I think it's should be doable, but beware, it would invalidate the real badge (but maybe you could re-sync it?).

rvalitov, 2023-07-25:
I tried latest firmware 0.87. I could read the VIGIK badges - that was working in the past releases, too. But emulation still does not work. I have a simple badge that does not have any internal counter or something like that.

AkechiShiro, 2023-07-25:
To be clear, on my testing on a VIGIK reader, I've noticed that : 
- Detect field (new Applications/Tools/"NFC/RFID detector released on the 0.86 release) => detect field, showed very promptly the frequency of the VIGIK reader and then the emulation was probably canceled by a HALT command.
- After 0.87, trying "detect field", shows that even if HALT gets sent, it does not cancel the emulation "forever".

- My badge has an anti-protection counter however I have 2 VIGIK readers, both V2 : 
  - Reader A : Increments counter (works only one time use, the increment counter fails and that desynchronizes the original badge)
  - Reader B : Does not increment counter (tested working with FP Zero under firmware 0.87)

I have found other reader (not VIGIK) still using Mifare Classic, that do not work, I'll investigate later why.

u1735067, 2023-08-01:
Like others, emulation seems ok on Vigik v2 now with 0.87.0, but reading through MCT on Android is very random.

Also, to recall some informations about Vigik (you can find more at http://2014.hackitoergosum.org/slides/day3_A_common_weakness_in_RSA_signatures:extracting_public_keys_from_communications_and_embedded_devices_Renaud_Lifchitz_hes2014.pdf starting slide 21), you have 2 main types of badges:
* **service badges**, used by official services like mentioned in previous comments, that contains a standardized message with the (short) expiration date, valid opening hours, ... And this message is RSA signed so it cannot be tampered. Technically you can clone the badge (as far as I understand, the only protection is the RSA signature, no counter), but it won't last more than the few days of the original badge ; also see the warning at https://github.com/flipperdevices/flipperzero-firmware/issues/1345#issuecomment-1493345464 about cloning them, because it might not be legal (they're not personal use)
* **residential badges**, containing manufacturer specific data (or not, just UID)
  * some use a counter to detect copies (see #1345 for more details, also if detected your badge might be blocked) ; this counter might not always be incremented (like for some parking access)?
  * some just use static data and can be copied just fine (unless they can detect magic/modifiable badges) ; beware however that the counter check can sometimes only be disabled and could be enabled later if the building manager request it

danila115, 2023-08-31:
Hi! I met with the problem that the mifare classic 1k label is read and emulated successfully, but when writing to a disc (mifare zero), it says that "the type of card is not correct." For the intercom key, it will be enough to copy only the UID, but I can’t do it in any way. 
The description of the information read from the tag corresponds to the key read from the original (ISO 14443-3 NFC-A)

theblackhole, 2023-08-31:
> Hi! I met with the problem that the mifare classic 1k label is read and emulated successfully, but when writing to a disc (mifare zero), it says that "the type of card is not correct." For the intercom key, it will be enough to copy only the UID, but I can’t do it in any way. The description of the information read from the tag corresponds to the key read from the original (ISO 14443-3 NFC-A)y

Hi did you use the "Magic" app on flipper to write to your badge ? If it doesn't work, it could be a magic gen2 badge which is not supported by flipper but works with Mifare Classic Tool on Android.
In any case, it is not related to this issue and you should ask your question [on the forum](https://forum.flipper.net/) if it has not been already asked ;)

doomwastaken, 2023-11-03:
@rvalitov @goutchye @BestPig @Pingvin0 could you please verify it with latest release firmware? There are was an entire NFC refactor

BestPig, 2023-11-03:
Upgraded to 0.94.1 and reading the emulated tag with my Android phone seems stable now.
Will try on a reader that never worked on previous version in some hours :pray:

Pingvin0, 2023-11-03:
> @rvalitov @goutchye @BestPig @Pingvin0 could you please verify it with latest release firmware? There are was an entire NFC refactor

The firmware works perfectly for my use-case now. Don't know about the rest of the issues reported in the comments

Hugal31, 2023-11-04:
It now works every time with the Classic Mifare Tools app, but still no dice with my Vigik reader v1. Great job though!

BestPig, 2023-11-04:
I also tried on a reader that never worked (I don't have the brand of the reader as it's built-in in a machine), but also not working. There is no react of the reader.

rvalitov, 2023-11-04:
Can't check, temporarily out of country, sorry.

AkechiShiro, 2023-11-04:
Picture of your reader ? @BestPig and brand please ?

fridgepoet, 2023-11-05:
I have a badge that was read and emulated and works on a Vigik V2 reader on v0.93.0.
On v0.94.1, the emulation has stopped working on this same reader.

Is there any information I can provide to help out?

theblackhole, 2023-11-05:
> I have a badge that was read and emulated and works on a Vigik V2 reader on v0.93.0. On v0.94.1, the emulation has stopped working on this same reader.

Same here, I tried with a previous clone (v2 file) and a fresh clone after the update (v4), both don't work anymore on a Vigik V2 reader (Urmet) as if the #2825 patch no longer works with this big NFC refactoring.

doomwastaken, 2023-11-21:
Thank you for details, we will look into this shortly

doomwastaken, 2023-11-22:
@theblackhole could you please show me the exact model (or at least a photo) of the Vigik V2 reader you are referencing, we will try to order on for testing

theblackhole, 2023-11-22:
@doomwastaken Of course! Not 100% sure but I think it's a [Urmet T25VK2](https://www.urmet.fr/t25vk2+lecteur+vigik+residents+format+t25+t25vk2-4904). It has been installed between 2019 and 2020.

<details>
  <summary>And here's the intercom for context</summary>
  [image]
   [image]
</details>

goutchye, 2023-12-08:
Hi @doomwastaken 

Sorry for the delay of my answer. 
I updated my Flipper today, with the last firmware available and the emulation still doesn't work for me.
I dump a legitime vigik tag and tried on a vigik reader (comelit UG1) without success, and moreover I tried on a Pixel 4a with the MFC app and the app didn't findthe tag, it didn't even find any UID.
Sorry

[9 more replies]