flipperdevices/flipperzero-firmware #1469: Newest NFC dev update break some reading

V0lk3n, 2022-07-27:
### Describe the bug.

In the previous version i was able to Read the NFC of Skylanders figures.

Since the new dev update of 27 jully, when i try to read it on the new NFC reader it keep saying "Reading card... Don't Move" for ever.

So i'm not able to read my Skylanders and save the signal.

### Reproduction

1. Press NFC
2. Read Card
3. It read card for ever without returning result

gornekich, 2022-07-27:
Could you please send logs here.
1) Open settings->system->log level = Debug
2) Connect Flipper to USB. Open serial port to access CLI, print logs
3) Read card and send logs here

V0lk3n, 2022-07-27:
I'm trying but unfortunately the log say nothing while it said "reading card dont move".

 The log didnt return result since i press "Read" on the flipper zero. Nothing happen from "detecting NFC card" to "Reading card dont move".

gornekich, 2022-07-27:
Are you shore that you set log level to Debug? Can you send logs from CLI starting before you press Read on flipper

V0lk3n, 2022-07-27:
I got result in "Trace" mode only :

````
# Part "Detecting NFC Card"
1012163 [T][FuriHalNfc]: State change 0 -> 10
1012289 [T][FuriHalNfc]: State change 10 -> 20
1012462 [T][FuriHalNfc]: Timeout
1012565 [T][FuriHalNfc]: State change 0 -> 10
1012691 [T][FuriHalNfc]: State change 10 -> 20
1012864 [T][FuriHalNfc]: Timeout
# Put the Skylanders on the flipper "Reading Card... Don't Move"
1012967 [T][FuriHalNfc]: State change 0 -> 10
1013101 [T][FuriHalNfc]: State change 10 -> 11
1013126 [T][FuriHalNfc]: State change 11 -> 13
1013131 [T][FuriHalNfc]: State change 13 -> 30
1013235 [T][FuriHalNfc]: State change 0 -> 10
1013369 [T][FuriHalNfc]: State change 10 -> 11
1013394 [T][FuriHalNfc]: State change 11 -> 13
1013402 [T][FuriHalNfc]: State change 13 -> 30
1013506 [T][FuriHalNfc]: State change 0 -> 10
1013635 [T][FuriHalNfc]: State change 10 -> 11
1013660 [T][FuriHalNfc]: State change 11 -> 13
1013665 [T][FuriHalNfc]: State change 13 -> 30
1013769 [T][FuriHalNfc]: State change 0 -> 10
1013898 [T][FuriHalNfc]: State change 10 -> 11
1013923 [T][FuriHalNfc]: State change 11 -> 13
1013928 [T][FuriHalNfc]: State change 13 -> 30
1014032 [T][FuriHalNfc]: State change 0 -> 10
1014162 [T][FuriHalNfc]: State change 10 -> 11
1014187 [T][FuriHalNfc]: State change 11 -> 13
1014192 [T][FuriHalNfc]: State change 13 -> 30
1014296 [T][FuriHalNfc]: State change 0 -> 10
1014428 [T][FuriHalNfc]: State change 10 -> 11
1014453 [T][FuriHalNfc]: State change 11 -> 13
1014458 [T][FuriHalNfc]: State change 13 -> 30
1014562 [T][FuriHalNfc]: State change 0 -> 10
1014692 [T][FuriHalNfc]: State change 10 -> 11
1014717 [T][FuriHalNfc]: State change 11 -> 13
1014722 [T][FuriHalNfc]: State change 13 -> 30
1014826 [T][FuriHalNfc]: State change 0 -> 10
1014956 [T][FuriHalNfc]: State change 10 -> 11
1014981 [T][FuriHalNfc]: State change 11 -> 13
1014986 [T][FuriHalNfc]: State change 13 -> 30
1015090 [T][FuriHalNfc]: State change 0 -> 10
1015220 [T][FuriHalNfc]: State change 10 -> 11
1015245 [T][FuriHalNfc]: State change 11 -> 13
1015250 [T][FuriHalNfc]: State change 13 -> 30
```

V0lk3n, 2022-07-27:
In Debug log level i got this result 

```

1256621 [I][LoaderSrv]: Starting: NFC
1256629 [I][AnimationManager]: Unload animation 'L1_Read_books_128x64'
1257721 [D][DolphinState]: icounter 15, butthurt 0
# It didnt return anything more
```

GMMan, 2022-07-27:
Skylander is [Mifare Classic](https://nfc.toys/#skylanders-and-mifare), so may still be subject to MFC emulation issues.

V0lk3n, 2022-07-27:
If that can help to know why it didnt read anymore, i post one saved signal of a skylanders figure with the previous firmware.

```
Filetype: Flipper NFC device
Version: 2
# Nfc device type can be UID, Mifare Ultralight, Mifare Classic, Bank card
Device type: UID
# UID, ATQA and SAK are common for all formats
UID: 94 AF 09 20
ATQA: 01 0F
SAK: 01
```

GMMan, 2022-07-27:
Is that the whole file? It's only UID emulation, so it probably never worked in the first place.

gornekich, 2022-07-27:
I see the reason now.
We don't support this ATQA and SAK compination, that's why previously we saved just UID part.
I will add fallback to read UID part for unrecognized cards in new nfc design tomorrow and flipper will read the card just as it read it before.
Further we need to verify that it's Mifare Classic and try to read sectors.
Thanks for information!

V0lk3n, 2022-07-27:
@GMMan It worked ^^ There is just a bit more job to do with the UID after that.

Also i dont think it's Mifare Classic. I invite you to look the documentation bellow that can really help you to do something i think @gornekich

https://nfc.toys/interop-sky.html

https://nfc.toys/data-giants.html

V0lk3n, 2022-07-27:
If i can be of any help let me know :) And thanks for you'r help and for adding the fallback to read UID :)

V0lk3n, 2022-07-27:
"Skylander toys in particular use the Mifare Classic standard, which divides each tag’s storage into sectors and blocks."

So it seem to bee Mifare Classic Standard, thats right.

Source : http://con-mod.com/skylanders-nfc/

GMMan, 2022-07-27:
It's close enough to a Mifare Classic that you might as well call it a Mifare Classic, ATQA/SAK not withstanding

V0lk3n, 2022-07-27:
> It's close enough to a Mifare Classic that you might as well call it a Mifare Classic, ATQA/SAK not withstanding

As i said "thats right" so i give you the point, i dont know where you want to go but well..

V0lk3n, 2022-07-28:
Seem like you fixed it! And now its better :) thanks!

[image]

closed 2022-07-28 by V0lk3n