flipperdevices/flipperzero-firmware #1472: NFC: Emulating a UID of a previously saved ISO-14443A-4 does not work because SAK tells there should be an ATS

doegox, 2022-07-27:
### Describe the bug.

Flipper allows to emulate the UID of cards it can't fully emulate, such as a MIFARE DESFire.
But it emulates the recorded SAK as well.
The SAK of an ISO-14443-4 has bit6=1 (e.g. SAK=20), contrarily to ISO-14443-3 cards such as MIFARE Classic, NTAG, Ultralight, with a SAK with bit6=0.
The problem is that when emulating a UID, the Flipper emulates an ISO-14443-3 without an ATS, but the reader seeing a SAK=20 is waiting for the ATS that never comes.

### Reproduction

Steps:
- read an ISO-14443A-4, for example a MIFARE DESFire.
- emulate the UID of the saved card
A reader will fail detecting the emulated card.

### Target

Tested on today Release and Development firmwares

### Anything else?

Two possible solutions:
1) record and save the ATS in ISO-14443A-4 dumps, then when "emulating UID", replay the ATS in the anticollision.
2) clear bit6 of SAK when "emulating UID"

To make it robust, they can be combined:
- record and save the ATS in ISO-14443A-4 dumps
- "emulating UID" : check that if bit6 of SAK=1, then check if ATS is available in the saved file. If not, clear SAK bit6 during the emulation

gornekich, 2022-07-28:
Hello! Thanks for bug report and solution.
Now I will add quick fix to clear 6 bit in SAK in UID emulation. Then we will save ATS and follow your suggestions

skotopes, 2022-09-06:
Please check latest release and reopen if issue persist

closed 2022-09-06 by skotopes