flipperdevices/flipperzero-firmware #1473: sub-ghz capture file format exporter/converter

ResistanceIsUseless, 2022-07-28:
### Description of the feature you're suggesting.

Would it be possible to get a tool that converts the raw data from a capture into hex or something that could be easily converted to other tools? For example, it would be nice to be able to transfer captures between [rfcrack ](https://github.com/cclabsInc/RFCrack)and flipper.

skotopes, 2022-07-28:
Hi. Flipper raw format is basically demodulated signal durations in microseconds. Tools that works with I/Q data are not going to work with it.

ResistanceIsUseless, 2022-07-28:
Just to clarify, it's basically the waveform plotted into ascii? If so the only thing i can think of that might work is universal hacker radio.

The only thing i could find that does similar raw files is this. http://www.rfxcom.com/WebRoot/StoreNL2/Shops/78165469/MediaGallery/Downloads/RFXtrx_User_Guide.pdf

Which interestingly enough has a bunch of device signals decoded.

Closing out, but I'll come back and add a comment if i find a way to transfer captures between devices.

closed 2022-07-28 by ResistanceIsUseless