flipperdevices/flipperzero-firmware #1476: NFC Reading wrong bank card number

VladichaYT, 2022-07-28:
### Describe the bug.

Hey all, when i try to read a bank card, it gives me different bank card code. For example if my card ends with 1234, and i read it with the flipperzero it  will show totally different code. Same thing with a different card. 

### Reproduction

1. NFC
2. Read card
3. Reads card, shows the bank
4. Click run
5. Shows the wrong bank card code, exp, UID...

skotopes, 2022-07-28:
Hi. Which bank it is? Also it looks like intended behavior: card number used for EMV doesn't have to be same.

VladichaYT, 2022-07-28:
> Hi. Which bank it is? Also it looks like intended behavior: card number used for EMV doesn't have to be same.

Raifaisen bank, visa. Oh okay, because a guy said that when he reads his bank card it shows the real numbers.

closed 2022-12-01 by skotopes