flipperdevices/flipperzero-firmware #1479: Replaying captured CAME TWEE sub-Ghz key crashes the Flipper Zero

korfuri, 2022-07-28:
Hi,

I'm running the release Firmware 0.62.1. I captured the following key in the wild: https://dev.flpr.app/s/#path=subghz/Test1.sub&Filetype=Flipper+SubGhz+Key+File&Version=1&Frequency=433920000&Preset=FuriHalSubGhzPresetOok650Async&Protocol=CAME+TWEE&Bit=104&Key=83+BD+00+FA+00+B7+00+BA I don't know the actual source of the signal, I got this by chance when playing around.

It shows up as a CAME TWEE key, I can save it, but pressing "send" on the "Emulate" screen crashes the Flipper Zero, requiring a hard reset to recover.

I'm happy to provide any other info that may be needed.
Thanks!

skotopes, 2022-07-28:
Hi, please check Release Candidate

Skorpionm, 2022-07-28:
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: CAME TWEE
Bit: 104
Key: 83 BD 00 FA 00 B7 00 BA

Bit: 104 this is an invalid value for the number of bits, it was fixed in version 0.63.2. check again

korfuri, 2022-07-28:
Emulating that file still crashes with RC 0.63.2.

I am not able to recapture the same signal so I can't tell whether it would be captured differently under RC 0.63.2.

skotopes, 2022-07-28:
I think I know what's wrong. Wait for next RC.

skotopes, 2022-09-06:
Please check latest release and reopen issue if problem persist

closed 2022-09-06 by skotopes

korfuri, 2022-09-15:
I confirm that with release 0.66.1 I can't open the file to replay it. Thanks!